billw2/pikrellcam #2: Security aspects

Yves911, 2015-07-21:
I discovered your software few days ago and i have been very impressed about how its powerful (live stream with good quality and reliable motion detection). I was using motion until now and consider to switch to your software.
I would like to open the stream on internet but with some restrictions and security:
1. do you plan to provide instructions to make the server running on https? (i can still use stunnel to have https)
2. do you have plan to enhance security with a username/password (currently if you connect to 'http://_______fr:8081/mjpeg_read.php directly you will see the image without the need to enter any crendetial'

Thanks for this great work !

Yves911, 2015-07-21:
using this guide i managed to enhance security : https://www.digitalocean.com/community/tutorials/how-to-set-up-http-authentication-with-nginx-on-ubuntu-12-10

mkoryak, 2015-07-22:
also, you should consider fixing this bug:

http://hostname/media.php?dir=media/videos/../../../../../../../../etc&file=

lets you easily look at any file in your filesystem, but yeah, basic auth should be used

billw2, 2015-09-26:
Password setting uses htpasswd.

closed 2015-09-26 by billw2