billw2/pikrellcam #9: Media directories empty in web interface

ektus, 2016-01-10:
I've got pikrellcam running on a RPi 2 with Raspbian Wheezy. The media directory is located on a 1TB 2,5" USB hard disk (FAT32). Motion detection works very well, movies and thumbs are stored on the drive, but the web interface says "No files" and "Disk: 0.0B   Free: 0.0B   Used: 0.0BB (0.0 %)"

I'm running pikrellcam and motioneye in parallel, pikrellcam with the pi-cam and motioneye with a network cam. The drive is mounted statically with an entry in fstab. I can view the videos and thumbs by browsing a samba share of the drive, but not from within the pikrellcam web interface.

```
========================================================
2016-01-09 22:31:58 ===== PiKrellCam 2.1.8 started =====
========================================================
using FIFO: /home/pi/pikrellcam/www/FIFO
using mjpeg: /run/pikrellcam/mjpeg.jpg
22:31:58 : execl:[sudo chown pi.www-data /media/TOSHIBA_1TB/pikrellcam]
22:31:58 : execl:[sudo chmod 775 /media/TOSHIBA_1TB/pikrellcam]
22:31:58 : execl:[sudo chown pi.www-data /media/TOSHIBA_1TB/pikrellcam/archive]
22:31:58 : execl:[sudo chmod 775 /media/TOSHIBA_1TB/pikrellcam/archive]
22:31:58 : execl:[/home/pi/pikrellcam/scripts-dist/_init /home/pi/pikrellcam /media/TOSHIBA_1TB/pikrellcam/archive /media/TOSHIBA_1TB/pikrellcam /run/pikrellcam/mjpeg.jpg /home/pi/pikrellcam/www/FIFO /tmp/pikrellcam.log]
  www/media link is already set to /media/TOSHIBA_1TB/pikrellcam
  www/archive link is already set to /media/TOSHIBA_1TB/pikrellcam/archive
  /home/pi/pikrellcam/www/config.php: LOG_FILE not changed.
  /home/pi/pikrellcam/www/config.php: MJPEG_FILE not changed.
  /home/pi/pikrellcam/www/config.php: FIFO_FILE not changed.
  /home/pi/pikrellcam/www/config.php: PIKRELLCAM not changed.
  /home/pi/pikrellcam/www/config.php: VERSION not changed.
22:31:58 : execl:[/home/pi/pikrellcam/scripts/startup /home/pi/pikrellcam /media/TOSHIBA_1TB/pikrellcam /tmp/pikrellcam.log]
  MOUNT_DISK is not set.
22:31:58 : execl:[sudo chown pi.www-data /media/TOSHIBA_1TB/pikrellcam]
22:31:58 : execl:[sudo chmod 775 /media/TOSHIBA_1TB/pikrellcam]
22:31:58 : execl:[sudo chown pi.www-data /media/TOSHIBA_1TB/pikrellcam/archive]
22:31:58 : execl:[sudo chmod 775 /media/TOSHIBA_1TB/pikrellcam/archive]
22:31:58 : execl:[sudo chown pi.www-data /tmp/pikrellcam.log]
22:31:58 : execl:[sudo chmod 664 /tmp/pikrellcam.log]
22:31:58 : execl:[sudo chown pi.www-data /media/TOSHIBA_1TB/pikrellcam/videos]
22:31:58 : execl:[sudo chmod 775 /media/TOSHIBA_1TB/pikrellcam/videos]
22:31:58 : execl:[sudo chown pi.www-data /media/TOSHIBA_1TB/pikrellcam/thumbs]
22:31:58 : execl:[sudo chmod 775 /media/TOSHIBA_1TB/pikrellcam/thumbs]
22:31:58 : execl:[sudo chown pi.www-data /media/TOSHIBA_1TB/pikrellcam/stills]
22:31:58 : execl:[sudo chmod 775 /media/TOSHIBA_1TB/pikrellcam/stills]
22:31:58 : execl:[sudo chown pi.www-data /media/TOSHIBA_1TB/pikrellcam/timelapse]
22:31:58 : execl:[sudo chmod 775 /media/TOSHIBA_1TB/pikrellcam/timelapse]
22:31:58 : circular buffer allocate: 26.25 MBytes (35 seconds at 6.0 Mbits/sec)
22:31:58 : Server: error binding
22:31:58 : Server running...waiting for connections.
22:31:58 : MJPEG server is listening on port '9999'
sunrise/sunset times: valid  dawn/dusk times: valid
  dawn:    7:07
  sunrise: 7:37
  sunset:  17:08
  dusk:    17:38
22:31:59 : execl:[/home/pi/pikrellcam/scripts-dist/_log-lines 1000 /tmp/pikrellcam.log]

========================================================
2016-01-10 ================== New Day ==================
========================================================
sunrise/sunset times: valid  dawn/dusk times: valid
  dawn:    7:07
  sunrise: 7:37
  sunset:  17:09
  dusk:    17:39
00:00:00 : execl:[/home/pi/pikrellcam/scripts-dist/_log-lines 1000 /tmp/pikrellcam.log]
08:11:52 : Video record: /media/TOSHIBA_1TB/pikrellcam/videos/motion_2016-01-10_08.11.52_0.mp4.h264 ...
08:11:52 : event preview save: copy /run/pikrellcam/mjpeg.jpg -> /run/pikrellcam/motion_2016-01-10_08.11.52_0.jpg
08:11:59 : event preview save: copy /run/pikrellcam/mjpeg.jpg -> /run/pikrellcam/motion_2016-01-10_08.11.52_0.jpg
08:11:59 : event preview save: copy /run/pikrellcam/mjpeg.jpg -> /run/pikrellcam/motion_2016-01-10_08.11.52_0.jpg
08:12:29 : Video motion record stopped. Header size: 28  h264 file size: 5224742
08:12:29 :     first detect: direction  totals - direction: 4  burst: 0  max burst count: 0
08:12:29 : execl:[(MP4Box -quiet -tmp /tmp -fps 10 -add /media/TOSHIBA_1TB/pikrellcam/videos/motion_2016-01-10_08.11.52_0.mp4.h264 /media/TOSHIBA_1TB/pikrellcam/videos/motion_2016-01-10_08.11.52_0.mp4 2> /dev/null && rm /media/TOSHIBA_1TB/pikrellcam/videos/motion_2016-01-10_08.11.52_0.mp4.h264)]
08:12:29 : execl:[/home/pi/pikrellcam/scripts-dist/_thumb /run/pikrellcam/motion_2016-01-10_08.11.52_0.jpg /media/TOSHIBA_1TB/pikrellcam /home/pi/pikrellcam/www/FIFO /tmp/pikrellcam.log 60 88 43 385]
  convert /run/pikrellcam/motion_2016-01-10_08.11.52_0.jpg -crop 88x88+0+341 -resize 150x150\! /media/TOSHIBA_1TB/pikrellcam/thumbs/motion_2016-01-10_08.11.52_0.th.jpg
08:12:29 : event_preview_dispose(); removing /run/pikrellcam/motion_2016-01-10_08.11.52_0.jpg
```

billw2, 2016-01-10:
On Sun, 10 Jan 2016 00:44:21 -0800
ektus <notifications@github.com> wrote:

> I've got pikrellcam running on a RPi 2 with Raspbian Wheezy The media directory is located on a 1TB 2,5" USB
> hard disk (FAT32) Motion detection works very well, movies and thumbs are stored on the [...]

When you fstab mount, are you using umask=0000 ?

The web interface runs from user www-data and needs read/write permissions to be enabled
for it.  From linux (pikrellcam) permissions cannot be set on a FAT32 and permissions can
be set only when the drive is mounted.

Otherwise, could you format the partition you use for pikrellcam media as a linux partition?

ektus, 2016-01-10:
I'll have to look up the fstab line, but won't be able to do so prior to next Friday. The fstab line was a copy of the mounttab (sp?) line Raspian Wheezy generated automatically. I can try and format the partition to EXT4, but again not before Friday. I'll post again once I've tried that.

ektus, 2016-01-15:
fstab line is
```
/dev/sda1 /media/TOSHIBA_1TB vfat rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0077,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro 0 0
```

billw2, 2016-01-15:
With the fmask, dmask you have, the read/write permissions are not right.

I think it should work with:  fmask=0002,dmask=0002 or as a last
resort fmask=0000,dmask=0000

Bill

ektus, 2016-01-15:
okay, I'll try, but in the meantime I've got errors and the files system has been remounted read only :-(
error message: invalid cluster chain, seems to have happened just now as I was trying to delete some files via the Samba access from my PC.

ektus, 2016-01-15:
okay, changing fmask and dmask to 0002 has solved my problem. The web interface is working properly now. Thank You very much.

closed 2016-01-15 by ektus