billw2/pikrellcam #10: frame rates

MercedesArio, 2016-02-14:
Hi, I cross posted on raspberry.org forums but this may be a more appropriate place. Changing the video_fps setting in pikrellcam.config file has no effect on the video output. I output .h264 files at different video_fps settings (lower, higher than default 24) and examine them with "ffmpeg -i" (or VLC->Media Information) - the clips are always reported to be encoded at 25 frames per second.
Is there any way to change the video_fps value to produce a corresponding output?

Thanks!!!

billw2, 2016-02-14:
On Sun, 14 Feb 2016 12:15:11 -0800
MercedesArio <notifications@github.com> wrote:

> Hi, I cross posted on raspberry.org forums but this may be a more appropriate place. Changing the video_fps
> setting in pikrellcam.config file has no effect on the video output. I output .h264 files [...]

Oh, sorry I missed your forum post.  Raw h264 video is not in a container and so I don't
think there is frame rate information available.  So vlc plays back assuming a 25 rate.
I think you are recording at 30 fps and playing back at 25.
Put a stop watch on the video and compare to the annotated time on the video.

For example, I tested recording a h264 video at 30fps and timed 30 seconds of reported
annotate time playback with vlc.  The stopwatch time was 36 seconds which is right for a
30/25 fps ratio.   If you set video_fps really low, 10 or less, then the playback difference
is clearly apparent.  vlc plays it at 25 fps and you see a fast motion video.  I don't know
if vlc has an argument you can pass to tell it to use a fps.

This is why if you are boxing to mp4 video, there is a separate video_mp4box_fps
config which has to be specified and it should track the video_fps setting.

Bill

MercedesArio, 2016-02-17:
Hi Bill,

Sorry for my late reply. All is well!!! I did practically hundreds of tests to solve this mystery before realizing that all of this was done on a test Pi 2 that will be deployed in my vacation property in one of the Caribbean countries. Time ago I was advised to down-clock to: arm_freq_min=400; sdram_freq_min=200; core_freq_min=200 to make sure the gadget does not overheat in the hot climate. I never realized that this will affect the fps issue!!! But it did!!! I reverted the Pi to default clock frequencies and now am enjoying a stunning 1080p full 30fps output playing in a correct time frame. Would love to know the theory behind though, would you mind sharing your wisdom?

Thanks a lot and congratulations on your amazing project!!!

Mercedes