billw2/pikrellcam #14: Still image motion detect

Harrie1966, 2016-05-28:
I have a question I would like to use the Function timelapse but only when there is motion detected. Is this possible and I can customize this turn or a script. I'm very satisfied with this program works very well. only timelapse takes up too much space on an SD card. I hope you can help me.

Egregius, 2016-08-17:
What about this? 
on_motion_begin echo "tl_start 300" > ~/pikrellcam/www/FIFO

Haven't tested, don't know if the video file will also be created.

linktohack, 2016-10-07:
Hello, thank you for the awesome project. I'm having the same question as @Harrie1966. Actually I came to the same solution as @Egregius , but I want to have same ability as motion video, e.g. be able to capture with a `pre` event duration... Is there a way to achieve this?

Egregius, 2016-10-08:
That's something for a real programmer I guess...
Or try to go advanced scripting and store snapshots every second in memory and remove them is no motion is detected. 
If motion is detected copy them to disk and combine them to a timelapse. 
Will not be that simple I guess.

Siddyboy, 2018-04-27:
Hi, I would also like to do as the OP mentioned - take stills (or a time lapse sequence of stills), but only on motion detect. Did anyone ever get anywhere with this? Did anyone try Egregius idea?