billw2/pikrellcam #15: Event Trigger http

Tristan79, 2016-07-09:
Is it possible to have http triggers? That is notion is detected it opens an custom url?

The reason I ask is that I would like it to have it included in my home automation. Such that I can send myself a message when a motion happens (in a region).

The other reason is I would like to use the motion detection with regions to detect walk direction and zone (= room or part of room) occupation. And of course record everything.

https://www.domoticz.com/forum/viewtopic.php?f=35&t=8390&p=90277&hilit=motioneye#p90277
http://domoticz.com/forum/viewtopic.php?f=23&t=11908&p=85624&hilit=camera+zones#p85624

billw2, 2016-07-09:
On Fri, 08 Jul 2016 21:38:49 -0700
Tristan Crispijn <notifications@github.com> wrote:

> Is it possible to have http triggers? That is notion is detected it opens an custom url?
>
> The reason I ask is that I would like it to have it included in my home automation. Such that I can send [...]

It's a todo I am planning on because I have in mind being able to do motion tracking with servos.
I'm planning to have a real time status file or pipe of the motion event in progress updated with
position, magnitude and direction of each detection as it happens. This will be something a motion
begin script can monitor throughout a motion event and take action on as the event is in progress.

Right now all you can do is a post processing of the event because the file with the information may
not be flushed until the event ends.  You can get a csv file of the motion event by enabling motion_stats
in pikrellcam.conf and the file will be written alongside the video in the videos directory.  But this is
something for a motion end script to look at so is probably not what  you wanted.

Bill

Tristan79, 2016-07-10:
That would be very very great! The position, magnitude and direction of each detection will be most helpful in fine-tuning walk direction!

Egregius, 2016-08-17:
The new /run/pikrellcam/motion-events is awesome, thank you for that!
I altered a bit your motion-send-alarm2 script and came up with this:
`#!/usr/bin/env python
import sys
import socket
import time
import urllib2
holdoff = 3
filename = '/run/pikrellcam/motion-events'
file = open(filename,'r')
state = 'wait'
tsent0 = 0
tsent1 = 0
tsent2 = 0
tsent3 = 0
tsent4 = 0
tsent5 = 0
tsent6 = 0
tsent7 = 0
tsent8 = 0
tsent9 = 0
while 1:
	where = file.tell()
	line = file.readline()
	if not line:
		time.sleep(0.1)
		file.seek(where)
		continue
	line = line.strip('\n')
	if (line.find('<motion') == 0):
		state = 'motion'
	elif (line.find('</motion>') == 0):
		state = 'wait'
	elif (line.find('<end>') == 0):
		sys.exit(1)
	elif (state == 'motion'):
		if (line.startswith(str(0))):
			tnow = time.time()
			if (tnow >= tsent0 + holdoff):
				r, x, y, dx, dy, mag, count = line.split()
				if (int(mag) >= 12 and int(count) >= 20):
					urllib2.urlopen('http://192.168.0.10:8084/json.htm?type=command&param=switchlight&idx=545&switchcmd=On&level=0&passcode=').read(1000)
					tsent0 = tnow
		elif (line.startswith(str(1))):
			tnow = time.time()
			if (tnow >= tsent1 + holdoff):
				r, x, y, dx, dy, mag, count = line.split()
				if (int(mag) >= 12 and int(count) >= 20):
					urllib2.urlopen('http://192.168.0.10:8084/json.htm?type=command&param=switchlight&idx=545&switchcmd=On&level=0&passcode=').read(1000)
					tsent1 = tnow
		elif (line.startswith(str(2))):
			tnow = time.time()
			if (tnow >= tsent2 + holdoff):
				r, x, y, dx, dy, mag, count = line.split()
				if (int(mag) >= 12 and int(count) >= 20):
					urllib2.urlopen('http://192.168.0.10:8084/json.htm?type=command&param=switchlight&idx=545&switchcmd=On&level=0&passcode=').read(1000)
					tsent2 = tnow
		elif (line.startswith(str(3))):
			tnow = time.time()
			if (tnow >= tsent3 + holdoff):
				r, x, y, dx, dy, mag, count = line.split()
				if (int(mag) >= 12 and int(count) >= 20):
					urllib2.urlopen('http://192.168.0.10:8084/json.htm?type=command&param=switchlight&idx=545&switchcmd=On&level=0&passcode=').read(1000)
					tsent3 = tnow
		elif (line.startswith(str(4))):
			tnow = time.time()
			if (tnow >= tsent4 + holdoff):
				r, x, y, dx, dy, mag, count = line.split()
				if (int(mag) >= 12 and int(count) >= 20):
					urllib2.urlopen('http://192.168.0.10:8084/json.htm?type=command&param=switchlight&idx=545&switchcmd=On&level=0&passcode=').read(1000)
					tsent4 = tnow
		elif (line.startswith(str(5))):
			tnow = time.time()
			if (tnow >= tsent5 + holdoff):
				r, x, y, dx, dy, mag, count = line.split()
				if (int(mag) >= 12 and int(count) >= 20):
					urllib2.urlopen('http://192.168.0.10:8084/json.htm?type=command&param=switchlight&idx=545&switchcmd=On&level=0&passcode=').read(1000)
					tsent5 = tnow
		elif (line.startswith(str(6))):
			tnow = time.time()
			if (tnow >= tsent6 + holdoff):
				r, x, y, dx, dy, mag, count = line.split()
				if (int(mag) >= 12 and int(count) >= 20):
					urllib2.urlopen('http://192.168.0.10:8084/json.htm?type=command&param=switchlight&idx=545&switchcmd=On&level=0&passcode=').read(1000)
					tsent6 = tnow
		elif (line.startswith(str(7))):
			tnow = time.time()
			if (tnow >= tsent7 + holdoff):
				r, x, y, dx, dy, mag, count = line.split()
				if (int(mag) >= 12 and int(count) >= 20):
					urllib2.urlopen('http://192.168.0.10:8084/json.htm?type=command&param=switchlight&idx=545&switchcmd=On&level=0&passcode=').read(1000)
					tsent7 = tnow
		elif (line.startswith(str(8))):
			tnow = time.time()
			if (tnow >= tsent8 + holdoff):
				r, x, y, dx, dy, mag, count = line.split()
				if (int(mag) >= 12 and int(count) >= 20):
					urllib2.urlopen('http://192.168.0.10:8084/json.htm?type=command&param=switchlight&idx=545&switchcmd=On&level=0&passcode=').read(1000)
					tsent8 = tnow
		elif (line.startswith(str(9))):
			tnow = time.time()
			if (tnow >= tsent9 + holdoff):
				r, x, y, dx, dy, mag, count = line.split()
				if (int(mag) >= 12 and int(count) >= 20):
					urllib2.urlopen('http://192.168.0.10:8084/json.htm?type=command&param=switchlight&idx=545&switchcmd=On&level=0&passcode=').read(1000)
					tsent9 = tnow
`
It can call a different http request for each region. I'm planning on working on this further to have it more integrated in my Domoticz and notification system.