billw2/pikrellcam #18: Request: Support Adafruit ServoHat

Yaytay, 2016-10-16:
Hi,

I am using the Adafruit Servo Hat for controlling my servos, mainly because it lets me power them with a separate 6V supply.
It would be really convenient for me if PiKrellCam supported the Servo Hat.

Would you be interested in adding support for it?
Would you be interested in me hacking about with PiKrellCam to get it working with the ServoHat?

Jim

billw2, 2016-10-16:
On Sat, 15 Oct 2016 23:48:37 -0700
Yaytay <notifications@github.com> wrote:

> Hi,
>
> I am using the Adafruit Servo Hat for controlling my servos, mainly because it lets me power them with a
> separate 6V supply. It would be really convenient for me if PiKrellCam supported the [...]

Sure, it would make PiKrellCam better to have Servo Hat support,
so I would definitely be interested in your modifications if you
would add that in.

Thanks,
Bill

Yaytay, 2016-10-18:
Hi Bill,

Well, I've got it so it'll talk to my servos (I can manually move the 
servos using the arrow buttons), but I can't work out how to get it to 
automatically track.
What am I missing?
The attached image shows what the screen looks like.

To talk to the Servo Hat I need to use i2c, and the least impact, most 
guaranteed to be installed, way to do that seems to be using wiringpi.
So my change has added a dependency on libwiringPi, is that OK?
The alternatives are:
* Linux smbus library (headers not installed on my pi, so likely not on 
others).
* pigpiod (should be a clean swap, that's what I use from Java, but only 
on recent raspbians).
* Bit bashing (too much like hard work).

I think you'll be able to see my work in progress on github:
https://github.com/Yaytay/pikrellcam
(bit new to git, normally use mercurial :))

Jim

billw2, 2016-10-19:
I don't have an image attachment in the email I got and on the issues page I don't see it either...
Can you try again or send directly to me:
    billw-at-gkrellm.net
but replace the -at-.

I want to avoid adding those dependencies so would like to have a C interface to
the Servo Hat if possible.  A couple of years back I had an experimental board with the
PCA9685 and I wrote a C interface to it that could be used as a start.  I've since put that board
out of service and am actually out of town for at least a week and so can't get to
it to try  to revive it.  So, let me give you the C code I have and maybe you can
take a look to see if you can modify it to use for Servo Hat in PiKrellCam.  I'll attach
a tar file, maybe it will get through, if not, can I send it to you via email?

The program I used  to test the pca9685 library routines is part of a much bigger test
program so I'm not including it.  But it's pretty basic, make sure defines are right
or change the defines to use some config values, call piopub_pca9685_setup(), and then the
other pca functions.  But piopub refers to the experimental board I mentioned and those
references should be changed to something appropriate for pikrellcam.  There may be
other compile issues if moving the code to the pikrellcam source, I haven't tried any testing.
Also, pi_model() needs to be looked at.  The return value the pca9685 code expects I don't
think is what pi_model() in pikrellcam currently returns.  I didn't look closely at that yet.

Bill
[pca9685.tar.gz](https://github.com/billw2/pikrellcam/files/537609/pca9685.tar.gz)

billw2, 2016-10-19:
Oh, just noticed in the setup() routine, I was originally linking to glib, so g_new0()
needs to be changed.  It can be a cmalloc(), but it really doesn't need to be dynamic.  It
could just be a static declaration for pikrellcam purposes.

Yaytay, 2016-10-19:
Hi Bill,

That code looks simple enough - I gave up with using plain file handles 
when I wanted to read the MODE register, but you don't bother doing that :)
I'll replace my code with that and give it a go today, hopefully.

However, I've realised that I'm going to run into a problem - I have 
four other PWM channels running off the ServoHat, and I'm going to run 
into locking issues.
So what I actually want to do is run everything through pigpiod.
Given that we don't want extra dependencies, I was considering adding a 
further option to send the calls to pwm_width_func to a UDP endpoint - 
essentially giving a high speed callback mechanism to anything that 
wants to read it (a named pipe would be another option, but UDP has the 
advantage of throwing stuff away and not blocking).
I'll see if I can make this work, and if so it's up to you whether you 
want it or not - I'll keep it separate from the raw PCA9685 stuff.

Jim