billw2/pikrellcam #19: Request: use h264 as input stream

pieterpoorthuis, 2016-11-06:
I'm using pikrellcam with my RPi 3 and the RPi 8mp camera module. The pikrellcam software works perfect, but the RPi camera module performs really bad in low light conditions.
I have an excellent light sensitive camera with a Sony Exmor sensor at home that captures everything during the night, see [here](https://www.alibaba.com/product-detail/SIP-F19HGAP-1920X1080-60-50fps-H_60164689025.html?spm=a2700.7724838.0.0.gAqWFV). The software of this camera is awful, but it can create a H264 and H265 stream.

Would it be possible to feed the H264 stream of my external camera into pikrellcam? So pikrellcam can do the motion detection?

billw2, 2016-11-07:
On Sun, 06 Nov 2016 11:25:03 -0800
Pieter Poorthuis <notifications@github.com> wrote:

> I'm using pikrellcam with my RPi 3 and the RPi 8mp camera module. The pikrellcam software works perfect, but
> the RPi camera module performs really bad in low light conditions. I have an excellent light sensitive camera
> with a Sony Exmor sensor at home that captures everything during the night, see [here]
> (https://www.alibaba.com/product-detail/ -1920X1080-60-50fps-H_60164689025.html?spm=a2700.7724838.0.0.gAqWFV).
> The software of this camera is awful, but it can create a H264 and H265 stream.
> 
> Would it be possible to feed the H264 stream of my external camera into pikrellcam? So pikrellcam can do the
> motion detection?

The issue is that the Pi camera software driver interface has an added feature that h264 motion vector data can
be interleaved with h264 video data and pikrellcam uses those motion vectors to detect motion.  So pikrellcam
relies on direction and magnitude vectors and this data is an added feature provided by the Pi camera software
developers.

Your SIP-F19HGAP camera uses the ONVIF protocol which has camera level motion detection based on
luminance variation within cells.  Its motion detection algorithm is not vector based and the ONVIF standard
does not provide for getting the motion vectors that pikrellcam needs.  So, unfortunately, it would be a major
rewrite for pikrellcam to support ONVIF and without motion vector data pikrellcam motion detection would not
be at all the same.

pieterpoorthuis, 2016-11-14:
Thanks for the clear explanation. So the only path to a solution would be to interleave the motion vector data in the H264 stream from the SIP-F19HGAP camera.

I'm really interested in combining the best of both worlds. I have technical contacts at the SIP-F19HGAP manufacturer. 
Do you have contact details of the person who developed the H264 stream with motion vectors for the RPi camera?

marktheunissen, 2017-05-19:
@pieterpoorthuis did you get anywhere with this? I'm also interested.

pieterpoorthuis, 2017-05-20:
No. I contacted the SIP-F19HGAP manufacturer and the Sony Exmor manufacturer, but got no response.
If anybody knows a developer who has worked on H264+motion vectors, please let me know

ristomatti, 2019-06-09:
Does anyone know if there's any other camera/driver which would provide this? PikrellCam motion detection algorithm is superior to any other software solution I've seen on consumer priced products or freely available software in avoiding false alarms while still being able to detect even small insects (which is what I've often used it to). It seems there's been so little development on the Pi camera sector. The Chinese clones have "innovative" lense options which allow easy focal length adjustment but the sensor is always the same cheap one. It does work with PiKrellCam though but seems to often create more random noise triggering false alarms.

It would be great to get one that can provide normal colors during daytime but still be able to detect IR in the dark like two or my newer consumer grade IP cameras. I assume they incorporate a relay to toggle the IR filter on and orr(?).

ristomatti, 2020-08-09:
Reply to myself from future me. There are nowadays modules with automatic night/day mode like this one: https://www.amazon.com/dp/B07R4JH2ZV. However it's only based on the older/cheaper 5MP sensor.