billw2/pikrellcam #20: Rotation problem...

Tristan79, 2016-12-29:
I have a problem with rotation. I wanted to use this to monitor a door...  so I placed the camera like I though was the best view..., thinking I could rotate the image:

[image]

so i edited /home/pi/.pikrellcam/pikrellcam.conf and at the end I placed rotation 90 still thinking it would come out fine. The camera has not moved inbetween...

[image]

but... when it rotates, it cuts off the parts I want!? And adds new parts...  it keeps the same tv aspect ratio :-/

is it possible in any way such that the view in the first image (although good for now, but with a  90 degree crooked neck when viewing) is somehow rotated on the webpage (and recording) to 1080x1920?

c765f614, 2017-08-24:
Coming back to the above situation, I got almost the same problem. After a few days trying to solve it, no solutions found yet.

Problem is that the visible section is 1920x1080 (1080p) but I want a visibility of 1080x1920 (so the visibility 90deg rotated, so I can see what I need.
Is there or will there be a solution for this?

Many thanks..

creeperbait, 2019-04-03:
Yeah, I have EXACTLY the same situation; trying to capture the whole of a door, which tend to be, you know, verticalish.

cwdosman, 2019-04-03:
I just tried it on the fly, but when you change the resolution in /home/pi/.pikrellcam/pikrelcam.conf the video width to 1080 and the height to 1920, the preview was "verticalish". I didnt recorded a video but also there it should work.

CycleMark, 2019-04-14:
The settings you want are at the end of the file. Look for "rotation" 

[image]

Mark

creeperbait, 2019-05-16:
cwdosman, thanks! It's taken me a while to get to it, but that's exactly it, along with the corresponding still_width and still_height.