billw2/pikrellcam #22: Request: Face Recognition

tyborall, 2017-01-20:
At first, thank you for this great piece of software! There is no better one out there for Raspberry Pi. :-)

I would like to see face recognition in one of the future software versions. Would it be possible to implement this feature in the future?
I use your software on some Raspberries, one of it is for monitoring my automatic cat-flap and is making videos every time my cat leaves the home or comes back. As trigger I use a specific region of the cat-flap door, which only moves when the cat unlocks the door with it's implanted pet chip.
There are a few false positives (<2%) when it's very stormy outside and the door begins to flatter.
With face recognition, which should even work with a cat's face, I could eliminate the false positives.

[image]

Tristan79, 2017-01-20:
+1

I am using it the same way you do :-) "big cat brother"

And some way of tagging the facials, so you can search for it!

RawLiquid, 2017-01-25:
Next you would have to add an alarm for when a face is visible but not
recognized....
Just sayin, I know my dogs use their dog door (3 of 4 do anyways)...The
ones I want to know about are if a neighborhood cat has come in for a snack
and to wander around in the middle of the night....