billw2/pikrellcam #24: Depencency hell

zinahe, 2017-03-24:
Hi,

First of all, thank you very much for all the effort put forward in making pikrellcam and sharing it with the world. I tried to install it on a Pi3 running as headless server with very minimal linux (minibian) and I was shocked to find out that it downloaded about 200 libraries (close to 300MB in total) when I checked apt's log. See attachment for details.

However, when I graphed the dependency of the pikrellcam executable by itself (please see attached image), it doesn't look that complicated. Can somebody shade some light on what's going on ? I would still like to keep my pi running very lean on resources but don't know how to proceed. 

Would it be possible to run this great tool without the bloat that comes with it as part of running **install-pikrellcam.sh** ?

Regards,

Zinahe A.

[apt-log.txt](https://github.com/billw2/pikrellcam/files/869160/apt-log.txt)
[image]

billw2, 2017-03-25:
On Fri, 24 Mar 2017 13:48:07 -0700
Zinahe <notifications@github.com> wrote:

> Hi,
>
> First of all, thank you very much for all the effort put forward in making pikrellcam and sharing it with the
> world. I tried to install it on a Pi3 running as headless server with very minimal [...]

Seems like a lot of those packages should not be needed and maybe recommended packages
were installed in addition to just dependencies.  I'll have to test to see if adding the
--no-install-recommends flag to apt-get will make a difference.  If it doesn't I'm not sure why all of those
got installed.

Bill

Martinique, 2017-08-03:
The most likely culprit is libav-tools. I encountered the same bloat issue with ffmpeg, which it's based on, so I decided to instead compile it myself with the bits needed for Pi hardware acceleration. See this [blog post](https://maniaclander.blogspot.com/2017/08/ffmpeg-with-pi-hardware-acceleration.html). I don't know if PiKrellCam needs any of the omitted ffmpeg parts, though.

This could be fixed by making the libav-tools dependency optional and allowing the user to select another executable (ffmpeg) to run instead of _avconv_.

BTW, [Debian is dropping libav in favor of ffmpeg](http://news.softpedia.com/news/Debian-Moves-to-FFmpeg-and-Drops-Libav-484793.shtml), which is already available for Jessie in Backports and is the default multimedia framework in Stretch. Preparing PiKrellCam now will avoid a lot of hassle later, when Raspbian moves on to Stretch.