billw2/pikrellcam #26: Audio event?

maurogrsistemi, 2017-05-02:
First of all, the compliments for the good work.!!!

Then the question: is there a way to trigger a recording event when noise comes to a certain level?

Thanks

Mauro

billw2, 2017-05-03:
On Tue, 02 May 2017 06:57:45 -0700
maurogrsistemi <notifications@github.com> wrote:

> Then the question: is there a way to trigger a recording event when noise comes to a certain level?

For now there isn't, but I do plan to add a sound record event.

Bill

maurogrsistemi, 2017-05-04:
This is great!

 

Thanks

 

Mauro

billw2, 2020-06-05:
On Fri, 05 Jun 2020 13:40:54 -0700
kbmetz <notifications@github.com> wrote:

> Any updates on this? I'm running my Pi power off the GPIO pins and was able to add a USB microphone with no
> noise. It works great, but would be really nice to have this audio event feature
> 

Yes, actually I added that quite a while ago.  On the Help page, there is a link to audio trigger
documentation under "Recent Release Notes" for Version 4.1.3

Config is done in Setup->Config->Audio

When you adjust the Audio_Trigger_Level with the trigger enabled, a thin horizontal
bar will be drawn on the audio meter to show the trigger level.

kbmetz, 2020-06-06:
Bill, thank you! You are very correct! I did not realize it was there already! Sorry