billw2/pikrellcam #27: sound

arnauldb, 2017-05-28:
hello,

thank you for pikrellcam. I added a usb mic which is working and recording when I run 

> arecord -D plughw:1,0 -f cd test.wav

pikrellcam is recording a video when there is a motion, but there is no audio. Do I have to select an option or activate something somewhere ?

Thank you,

arnauldb, 2017-05-28:
ah, sorry. New icons appeared after reboot and I can toggle mic on/off and adjust volume.

Thank you for your work.

billw2, 2017-05-28:
On Sun, 28 May 2017 07:21:08 -0700
arnauldb <notifications@github.com> wrote:

> ah, sorry. New icons appeared after reboot and I can toggle mic on/off and adjust volume.
> 
> Thank you for your work.

OK, glad it works.  After upgrades pikrellcam needs to be restarted and web pages reloaded
to  pick up all changes.  That should make new stuff appear without having to do a reboot.

arnauldb, 2017-05-30:
yes, it works fine. I use it to record the life in my garden, birds, squirrels...thanks again for this. The difficult thing was, for me, to find a working system with video AND audio, to put in the pi. I still have to fine tune parameters but it works : [https://www.facebook.com/arnauld.bordes/posts/1479778918751586](url)