billw2/pikrellcam #28: sound quality.

arnauldb, 2017-06-22:
Hello,

I've been using pikrellcam for two weeks, I use it to record squirrels and birds in my garden, and it is working great. I'm wondering if there is a way to play with the sound quality, and how? Also, would you recommend any usb mic, as, for now, there is a lot of "noise" recorded in the video and I have to clean it in audacity to get a decent/clear sound?

Thank you,

arnauldb, 2017-07-06:
I found a good usb mic and the sound is good now.

closed 2017-07-06 by arnauldb

Yves911, 2017-07-08:
@arnauldb hi, can you share which mic are you using now with good results?

arnauldb, 2017-07-08:
>  can you share which mic are you using now with good results?

I bough the "CAD Audio U9 Condenser Microphone, Omni " on amazon: https://www.amazon.com/gp/offer-listing/B004P1BQG2/ref=dp_olp_all_mbc?ie=UTF8&condition=all

It is a usb mic, the results are good. To further reduce noise I also found out that it is better not to connect it directly to the raspberry pi (vibrations, ...?) but to use a short "extension cable USB 2.0, type A M/F" between the pi and mic: https://www.amazon.fr/Lindy-Rallonge-USB-2-0-noir/dp/B01LY6BGTC/ref=dp_ob_title_ce

Also, the mic has an illuminated red ring, which may be annoying....

The result: 

[image]

Yves911, 2017-07-08:
Thanks, i just bought this one https://www.amazon.fr/gp/product/B01LCIGY8U/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1 
I don't expect too much from it... but let's see what i can achieve with this cheap mic.