billw2/pikrellcam #29: night vision.

arnauldb, 2017-07-06:
Hello,

when using a night vision system, this camera: Raspberry Pi Camera Board - Night Vision (5MP), https://www.modmypi.com/raspberry-pi/camera/camera-boards/raspberry-pi-camera-board-night-vision-5mp and when motion detection is on, I have a lot of false detections and it records even when there is nothing moving. I guess it is due to the IR leds, but I am not sure. Do you have any clue or advice for this?

here is a screencast I did when motion is on: https://youtu.be/TJ1LpKHq220

Thank you,

billw2, 2017-07-06:
Under some lighting the camera generates a lot of noise vectors and the IR level may be
at the problem level.  If you turn on the display of "Vectors" you can see the noise and that
can give a good idea of the cause of the detects.

The light level during the video seems to fluctuate and that doesn't help.  When you look at the
vectors, you may see them clustered around the vertical line of the wall corner and it may
help to set motion regions that do not cross straight line divisions of light like that corner.

Bill

arnauldb, 2017-07-06:
"When you look at the
vectors, you may see them clustered around the vertical line of the wall corner and it may
help to set motion regions that do not cross straight line divisions of light like that corner." 

that was right, I moved the camera and no more noise vectors, I should have tried this before. Here is my new test: https://youtu.be/ul-UgSd2QQ8 

Thank you for the advice and PikrellCam.

closed 2017-07-06 by arnauldb