billw2/pikrellcam #33: Install on Raspbian Stretch 9.1 fails

pdfruth, 2017-10-24:
Trying to install pikrellcam on the latest version of Raspbian (Stretch 9.1).
The install-pikrellcam.sh script tries to install php5 (among other pre-requisites) which has been obsoleted and removed from the Stretch repositories (see https://wiki.debian.org/PHP#PHP_and_Debian).
Running install-pikrellcam.sh on a Raspberry Pi running the latest Raspbian Stretch 9.1 results in the following output messages;
--------- snip --------
```
Starting PiKrellCam install...
Installing packages:  gpac php5 php5-common php5-fpm nginx libav-tools bc sshpass mpack imagemagick apache2-utils libasound2-dev libmp3lame-dev
Running: apt-get update
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Get:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease [15.0 kB]
Fetched 15.0 kB in 2s (5,110 B/s)                         
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package php5 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package php5-common is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package php5-fpm is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'php5' has no installation candidate
E: Package 'php5-common' has no installation candidate
E: Package 'php5-fpm' has no installation candidate
./install-pikrellcam.sh: line 137: bc: command not found
./install-pikrellcam.sh: line 140: [: -gt: unary operator expected
Making /usr/local/bin/pikrellcam link.
Adding to /etc/sudoers.d: www-data permission to run pikrellcam as user pi:
./install-pikrellcam.sh: line 250: htpasswd: command not found
chown: cannot access 'www/.htpasswd': No such file or directory
grep: /etc/nginx/nginx.conf: No such file or directory
Turning off nginx access_log.
sed: can't read /etc/nginx/nginx.conf: No such file or directory
./install-pikrellcam.sh: line 266: [: -gt: unary operator expected
Installing /etc/nginx/sites-available/pikrellcam
    nginx web server port: 80
    nginx web server root: /home/pi/pikrellcam/www
cp: cannot create regular file '/etc/nginx/sites-available/pikrellcam': No such file or directory
sed: can't read /etc/nginx/sites-available/pikrellcam: No such file or directory
realpath: /etc/nginx/sites-enabled/default: No such file or directory
Changing /etc/nginx/sites-enabled/default link to pikrellcam
ln: failed to create symbolic link '/etc/nginx/sites-enabled/default': No such file or directory
Adding /etc/nginx/sites-enabled/default link to sites-available/pikrellcam.
ln: failed to create symbolic link '/etc/nginx/sites-enabled/default': No such file or directory
A password for the web page is not set.
sed: can't read /etc/nginx/sites-available/pikrellcam: No such file or directory
Failed to restart nginx.service: Unit nginx.service not found.

Install finished.
This install script does not automatically start pikrellcam.
To start pikrellcam, open a browser page to:
    http://your_pi
and click on the "System" panel and then the "Start PiKrellCam" button.
PiKrellCam can also be run from a Pi terminal for testing purposes.
```
--------- snip --------

Stretch repositories do include PHP7.0
Is pikrellcam compatible with PHP7.0?
Can the installer script be updated to detect Raspbian Stretch 9+ and install prerequisite packages '**php7.0 php7.0-common php7.0-fpm**' instead of '**php5 php5-common php5-fpm**'?

jrsGH, 2017-10-27:
Hi All,
Yes, same issue for me : this php issues makes the installation impossible on the latest Debian Jessie on the Raspberry_pi Zero W.
I wish I could install it since I like Pikrellcam for its power and interface
When can we expect a solution ?
Rgds

jrsGH, 2017-10-27:
Hi All,

This has been solved with a new release.
Just redo the clone and that's it !
My installation went nicely with hph7
Congrats to Pikrellcam team : this program is so powerfull and you are so reactive !
Rgds

pdfruth, 2017-10-29:
yup,
The changes to install-pikrellcam.sh in the latest commit look good.
Closing this issue.
Thanks

closed 2017-10-29 by pdfruth