billw2/pikrellcam #34: Batch downloading videos?

MRWhitehead, 2017-11-16:
I'm looking for a way to download batches of videos at the same time? The only download method I see is to download single individually selected videos at a time... Any suggestions appreciated!

ocdaochauphi, 2018-01-31:
You can use paragon software for win to read the SD card, then you can copy all videos at the same time. It is very fast.
By the way, could you manual record over 2 hours? I can not do this.

cima, 2018-05-11:
Sometimes it is not an option to remove the SD card or USB drive. It might be due to requirement of uninterupted operation or inconvenient acces to it. Download of all selected files at once is relevant usecase. As the raspberry has limited computational power and also limited space the method of streaming of on the fly constructed archive (TAR or ZIP) is exactly what will allow to tunnel multiple file through single HTTP request. A github project https://github.com/barracudanetworks/ArchiveStream-php has a very promissing library doing exactly what might make the feature easy to implement while acheiving the maximum effectivity and performance.

ristomatti, 2019-06-09:
This would be great. Even an option to disable autoplay would help a bit in my case. I often find myself having to download the recordings from Pi's through connections which are too slow for real time playback. But I'm tech savvy enough to add another web app to access the folders and do something like @cima suggested if I just could find the time from all my various other projects. :roll_eyes: At least this is something you can work around quite easily.

OT: Besides, I think PiKrellCam would deserve a completely new UI layer which could or even should be done by someone else. @billw2 has already added the FIFO command layer which to my understanding provides access to most of the features. I'm not sure if for example archiving and such would work.