billw2/pikrellcam #36: Don't get any image --> permissions problem?

bertmelis, 2017-12-13:
After a successful installation on a fresh Raspbian Stretch (Lite) on a RPI1B, Pikrellcam starts nicely. However, I don't get any image. I've found the following in the logs:
```
make_dir(/run/pikrellcam) failed. No such file or directory
Failed to create media directories or FIFO, exiting!
```
The camera module works as `raspistill` generates an image.

bertmelis, 2017-12-13:
ah wait, I installed it on my user account, and not `pi`. Is that supported?

billw2, 2017-12-13:
The user installing/running pikrellcam needs to be able to change owner/modes of directories so
that the web server www-data has permission to read/write files.  At least the user needs sudo
privilege and permission to access the /run directory.  Maybe you can set up another user
to work, but I haven't done it so can't give you the set of steps you would need.

bertmelis, 2017-12-13:
I always remove the `pi` user from the system and replaces it with my own with sudo and video privileges. But Pikrellcm won't display any image.

To troubleshoot I created the `pi` user again, reran the installation script from this user and now everything worked!

The thing I did different was that I had autostart at boot enabled the first time and I disabled it when I installed again with the `pi`-user.

bertmelis, 2017-12-13:
A quick search through the repo tells me there is a command executed by user `pi` on this line:
https://github.com/billw2/pikrellcam/blob/1697ed0ee687b7c592d9c4c40a4f291e86cebbf2/www/sys_command.php#L12

bertmelis, 2020-03-11:
This issue was resolved long time ago...

closed 2020-03-11 by bertmelis