billw2/pikrellcam #38: Config silently fails to load on start

churchmf, 2018-05-14:
Hello,

I'm attempting to setup on_motion_preview_save as explained in the configuration example however any changes I make to ~/.pikrellcam/pikrellcam.conf do not appear to be picked up upon restarting. 

I've tried adding an invalid line and it doesn't complain. I believe it may simply use the default config without any notification. 

Thanks for this awesome project and all the hard work!

Here's the output of running it with -v:
`root@3bc92cf78f95:~# /home/pi/pikrellcam/pikrellcam -v &
[1] 317
root@3bc92cf78f95:~# ======= PiKrellCam 4.1.6 started at 2018-05-14 04:35:21
== Dropped root priviledges-continuing as normal user ==
========================================================
  check_modes(/home/pi/pikrellcam/www) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
command FIFO: /home/pi/pikrellcam/www/FIFO
audio FIFO  : /home/pi/pikrellcam/www/audio_FIFO
mjpeg stream: /run/pikrellcam/mjpeg.jpg
make_dir(/home/pi/pikrellcam/media)
  make_dir(/home/pi/pikrellcam/media) dir already exists.
  check_modes(/home/pi/pikrellcam/media) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
make_dir(/home/pi/pikrellcam/media/archive)
  make_dir(/home/pi/pikrellcam/media/archive) dir already exists.
  check_modes(/home/pi/pikrellcam/media/archive) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
make_dir(/home/pi/pikrellcam/media/loop)
  make_dir(/home/pi/pikrellcam/media/loop) dir already exists.
  check_modes(/home/pi/pikrellcam/media/loop) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
04:35:21 :   execl[wait:1]: /home/pi/pikrellcam/scripts-dist/_init /home/pi/pikrellcam /home/pi/pikrellcam/media/archive /home/pi/pikrellcam/media /run/pikrellcam/mjpeg.jpg /home/pi/pikrellcam/www/FIFO /tmp/pikrellcam.log servos_off /home/pi/pikrellcam/media/loop
04:35:21 :   execl[wait:1]: /home/pi/pikrellcam/scripts/startup /home/pi/pikrellcam /home/pi/pikrellcam/media /home/pi/pikrellcam/media/archive /tmp/pikrellcam.log
  check_modes(/home/pi/pikrellcam/media) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
  check_modes(/home/pi/pikrellcam/media/archive) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
  check_modes(/home/pi/pikrellcam/media/loop) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
  check_modes(/tmp/pikrellcam.log) 664
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
  check_modes() (100644) execing: sudo chmod 664 /tmp/pikrellcam.log
04:35:21 :   execl[wait:1]: sudo chmod 664 /tmp/pikrellcam.log
make_dir(/run/pikrellcam)
  make_dir(/run/pikrellcam) dir already exists.
  check_modes(/run/pikrellcam) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
make_dir(/home/pi/pikrellcam/media/videos)
  make_dir(/home/pi/pikrellcam/media/videos) dir already exists.
  check_modes(/home/pi/pikrellcam/media/videos) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
make_dir(/home/pi/pikrellcam/media/thumbs)
  make_dir(/home/pi/pikrellcam/media/thumbs) dir already exists.
  check_modes(/home/pi/pikrellcam/media/thumbs) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
make_dir(/home/pi/pikrellcam/media/loop)
  make_dir(/home/pi/pikrellcam/media/loop) dir already exists.
  check_modes(/home/pi/pikrellcam/media/loop) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
make_dir(/home/pi/pikrellcam/media/stills)
  make_dir(/home/pi/pikrellcam/media/stills) dir already exists.
  check_modes(/home/pi/pikrellcam/media/stills) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
make_dir(/home/pi/pikrellcam/media/timelapse)
  make_dir(/home/pi/pikrellcam/media/timelapse) dir already exists.
  check_modes(/home/pi/pikrellcam/media/timelapse) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
make_fifo(/home/pi/pikrellcam/www/FIFO)
  make_fifo(/home/pi/pikrellcam/www/FIFO) FIFO already exists.
  check_modes(/home/pi/pikrellcam/www/FIFO) 664
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 664 already OK.
make_dir(/home/pi/pikrellcam/media/loop/videos)
  make_dir(/home/pi/pikrellcam/media/loop/videos) dir already exists.
  check_modes(/home/pi/pikrellcam/media/loop/videos) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
make_dir(/home/pi/pikrellcam/media/loop/thumbs)
  make_dir(/home/pi/pikrellcam/media/loop/thumbs) dir already exists.
  check_modes(/home/pi/pikrellcam/media/loop/thumbs) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
make_dir(/home/pi/pikrellcam/media/archive)
  make_dir(/home/pi/pikrellcam/media/archive) dir already exists.
  check_modes(/home/pi/pikrellcam/media/archive) 775
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 775 already OK.
make_fifo(/home/pi/pikrellcam/www/audio_FIFO)
  make_fifo(/home/pi/pikrellcam/www/audio_FIFO) FIFO already exists.
  check_modes(/home/pi/pikrellcam/www/audio_FIFO) 664
    getgrgid() current group name: www-data
    getpwuid() current user name: root
    User and group names already OK.
    Access mode 664 already OK.
04:35:21 : video circular buffer - 26.25 MB (35 seconds, 6.0 Mbits/sec)
top status area:    x:0 y:0 dx:800 dy:47
inform area:        x:0 y:47 dx:800 dy:354
adj_control area:   x:0 y:354 dx:800 dy:47
bottom status area: x:0 y:401 dx:800 dy:47
Event count down add [preset notify] count=22
04:35:21 : Server running...waiting for connections.
04:35:21 : MJPEG server is listening on port '9999'
sunrise/sunset times: valid  dawn/dusk times: valid
  dawn:    11:11
  sunrise: 11:37
  sunset:  1:18
  dusk:    1:44
04:35:23 :   execl[wait:1]: /home/pi/pikrellcam/scripts-dist/_log-lines 500 /tmp/pikrellcam.log
`

churchmf, 2018-05-14:
[Uploading pikrellcam.config…]()

churchmf, 2018-05-14:
Here's the config file in question: https://drive.google.com/file/d/1H3xSb-3GoZHoZkmWOPhcUREGwdNjYvqF/view?usp=sharing

timuritu, 2020-11-11:
Hello, I'm having the same problem, I think. Currently running ver 4.3.2. I created the pikrellcam.config file and loaded some "on_motion_begin" scripts but nothing.