billw2/pikrellcam #39: Display Additional Info On screen

cmedianu, 2018-06-02:
Hi Bill,

I'd like to implement the ability to display additional information on screen (sensor readings, etc.) , preferably by writing the info to the FIFO.

Any advice, before I get started, regarding where things should go?

Thanks!

billw2, 2018-06-02:
You can display info at the top to the left or right of the displayed date by sending annotate_string
commands to the FIFO.  This is the only way to display anything that also shows up on videos.

Search for annotate_string in the help page, and look at the ds18b20.py script in the scripts dir
for an example of displaying temperature data.  Scripts can free run and periodically write
annotate_string commands to the FIFO, or they can be run from cron or the at-commands.conf
file.

Also, here's a short script that demonstrates:

#!/bin/bash

echo "annotate_string prepend label test" > /home/pi/pikrellcam/www/FIFO

for i in 1 2 3 4 5
do
    x=$((i * 2))
    echo "annotate_string append sensor1 count1-$i" > /home/pi/pikrellcam/www/FIFO
    echo "annotate_string append sensor2 count2-$x" > /home/pi/pikrellcam/www/FIFO
    sleep 1
done

echo "annotate_string prepend label end test" > /home/pi/pikrellcam/www/FIFO
sleep 1
echo "annotate_string remove sensor1" > /home/pi/pikrellcam/www/FIFO
sleep 1
echo "annotate_string remove sensor2" > /home/pi/pikrellcam/www/FIFO
sleep 1
echo "annotate_string remove label" > /home/pi/pikrellcam/www/FIFO

cmedianu, 2018-06-02:
This works great, thanks!

closed 2018-06-02 by cmedianu