billw2/pikrellcam #43: Cleaning out every text from stream

tyborall, 2019-04-28:
I haven't found an option to remove every text from the stream. After removing any possible text is still states "Motion ON" and "REC (Stop)" in the left bottom corner and "1-1" in the right bottom corner. How can I remove those texts as well? If this is not possible, please move this issue to feature requests. It would be nice to even clean this out, either by option in config file or with bypass access method (e.g. something like mjpeg_stream_clean.php).

eribr, 2019-09-09:
I just made the switch from Motion Eye to PiKrellCam and was looking for the exact same feature. How do we get access to a "raw" mjpeg stream without the OSD.

eribr, 2019-09-11:
I did a quick hack to disable the OSD in the stream:

`diff --git a/src/config.c b/src/config.c
index 9177b9a..0f2d091 100644
--- a/src/config.c
+++ b/src/config.c
@@ -600,6 +600,10 @@ static Config  config[] =
          "#",
        "halt_enable",  "on", TRUE, {.value = &pikrellcam.halt_enable},  config_value_bool_set},

+       { "# Set to off to disable On Screen Display. To stream a clean image\n"
+         "#",
+       "osd_enable",   "on", TRUE, {.value = &pikrellcam.osd_enable},  config_value_bool_set},
+
        { "# Set to off to disable the multicast interface.\n"
          "#",
        "multicast_enable",     "on", TRUE, {.value = &pikrellcam.multicast_enable},  config_value_bool_set},
diff --git a/src/display.c b/src/display.c
index 930e39d..fd62328 100644
--- a/src/display.c
+++ b/src/display.c
@@ -142,6 +142,9 @@ static void
 i420_print(DrawArea *da, GlcdFont *font, int16_t color, int row,
                                int xs, int ys, int justify, char *str)
        {
+         if (!pikrellcam.osd_enable) {
+           return;
+         }
        int             x, y;

        if (row >= 0)
diff --git a/src/pikrellcam.h b/src/pikrellcam.h
index d915d2a..a39592a 100644
--- a/src/pikrellcam.h
+++ b/src/pikrellcam.h
@@ -796,6 +796,7 @@ typedef struct
        int             debug,
                        debug_fps;
        boolean halt_enable;
+       boolean osd_enable;
        }
        PiKrellCam;
`