billw2/pikrellcam #46: no audio from i2s-mems breakout

Sp4rkR4t, 2019-07-20:
I tried using a usb microphone to record audio and that worked but was extremely quiet as the mic was poor quality so I switched to an adafruit i2s-mems breakout, wired it all up, and tested it with arecord and it works perfectly with one exception pikrellcam does not record any audio with it, it opens the microphone for recording and the button appear on the UI but doesn't record anything.

billw2, 2019-07-24:
When a video is recorded with the mic on, look at the log for a record stop line and see if numbers
are given for audio frames and audio rate.  If you have good audio frames numbers it means data
was being read from the alsa interface.  If so is there any chance the audio could be muted when
pikrellcam opens the mic?  Run amixer or alsamixer while pikrellcam has the mic open and see
if anything about mute or capture level looks suspicious.

For the audio frames number, the log will have a line like:

Manual record stop (size:3.9MB < /tmp:275.4MB)  vid_time:7.75  vid_fps:24.00  audio_frames:371973  audio_rate:
48000

where the audio_rate times recorded seconds should be close to the audio_frames.
(eg 7.75 x 48000 close to 371973)

Also, what info do you get when you run:
   arecord --dump-hw-params /dev/null

This prints hw-params to the terminal and records audio into /dev/null so you have to ^C to quit
the record.

Bill

Sp4rkR4t, 2019-07-24:
When a recording ends the line in the log is like this;

`Manual record stop (size:4.1MB < /tmp:85949.0MB)  vid_time:5.50  vid_fps:24.01  audio_frames:131961  audio_rate:24000`

This is what I get from `arecord --dump-hw-params /dev/null`

`arecord --dump-hw-params /dev/null
Recording WAVE '/dev/null' : Unsigned 8 bit, Rate 8000 Hz, Mono
HW Params of device "default":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [32 64]
CHANNELS: 2
RATE: [8000 192000]
PERIOD_TIME: (166 8192000]
PERIOD_SIZE: [32 65536]
PERIOD_BYTES: [256 524288]
PERIODS: [2 4096]
BUFFER_TIME: (333 16384000]
BUFFER_SIZE: [64 131072]
BUFFER_BYTES: [256 524288]
TICK_TIME: ALL
--------------------
arecord: set_params:1339: Sample format non available
Available formats:
- S16_LE
- S24_LE
- S32_LE
`

billw2, 2019-07-24:
On Wed, 24 Jul 2019 03:53:21 -0700
Sp4rkR4t <notifications@github.com> wrote:

> When a recording ends the line in the log is like this;
>
> `Manual record stop (size:4.1MB < /tmp:85949.0MB)  vid_time:5.50  vid_fps:24.01  audio_frames:131961
> audio_rate:24000`
>
> This is what I get [...]
...

pikrellcam is opening the device and reading data, so all I can think to do is to verify
reading is from the correct device and that the set parameters are right.

From cat /proc/asound/cards or running arecord -l, make sure the card number is
right for the Adafruit card.  For example, for me, I would do this for my USB sound card like so:

$ cat /proc/asound/cards
 0 [ALSA           ]: bcm2835 - bcm2835 ALSA
                      bcm2835 ALSA
 1 [Device         ]: USB-Audio - USB PnP Sound Device
                      C-Media Electronics Inc. USB PnP Sound Device at usb-3f980

So my USB device is card 1 and so I will use plughw:1 and test run arecord with:

   $ arecord -d 2  -D plughw:1 -f S16_LE  -r 24000 test.wav

where I set the rate the same as you are using (24000) and set the format S16_LE which
pikrellcam uses and record for 2 seconds.

Make sure pikrellcam has the microphone closed and run the above arecord command and
verify there is sound in test.wav to check that you can record audio from the mic with the
same settings used by pikrellcam.

Next, open the microphone in pikrellcam and run the above arecord again.  It should error with:
   audio open error: Device or resource busy
and this verifies that pikrellcam tries to use the same device.

With the mic still open in pikrellcam, check the capture levels with aumix.  I get

$ amixer
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 1805 [3%] [on]
  Front Right: Playback 1805 [3%] [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch cswitch-joined
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 65536
  Front Left: Capture 65536 [100%] [on]
  Front Right: Capture 65536 [100%] [on]

which shows the "Capture"  device is "on" with capture level 100%.

To double check the Capture levels shown are for the right device, run alsamixer
and with the F6 key make sure the Adafruit mic is selected and then hit F5 to
show all controls.  Make sure all "Mic" indicators show 100% and are not muted
(mute is toggled with the "Space" key).  For my USB card I show two "Mic" level
bars and one of them can be muted and I have both set to 100%.

Sp4rkR4t, 2019-07-24:
Ok, when I use  `arecord -d 2  -D plughw:1 -f S16_LE  -r 24000 test.wav` the resulting file is silent as well so it must be the settings that are the issue, when I was testing the mic previously I was using `arecord -d 2 -D plughw:1 -r 48000 -f S32_LE test.wav` and that worked fine.

billw2, 2019-07-24:
The differences are the rate and format, so try:
   arecord -d 2  -D plughw:1 -f S16_LE  -r 48000 test.wav

If that works then set the pikrellcam rate the same.  Depending on if you have a
Pi1 or Pi2, set in pikrellcam.conf

audio_rate_Pi2 48000
and/or
audio_rate_Pi1 48000

If that does not work, then it must be format.  Unfortunately pikrellcam can't be configured to
a S32_LE format.  It would take adding some code to do that.  It would be trivial to set ALSA to that
but then the 32 bit .wav data has to be fed to lame to generate a .mp3 and offhand I'm not sure
how much effort that would take.

But I will say that the results of your arecord --dump-hw-params previously say that the Adafruit
driver claims it supports S16_LE.  If it does not then there is a bug there and maybe you could
report to them that arecord -f S16_LE is not working.  Or if it's the rate that's the problem, maybe
some other rate than 48000 would work, maybe try 22050.

Sp4rkR4t, 2019-07-24:
Tested it, the format is the issue I've reported this to adafruit, hopefully they can help. Thank you for helping me troubleshoot this device with your excellent software.

closed 2019-08-01 by Sp4rkR4t