billw2/pikrellcam #47: drift audio from video out of sync

AntonJouri, 2019-07-31:
When recording video with audio, the mic is ahead of the video, and it is getting worse every minute.
It is known, that the fps of Raspi-Camera and USB-Microphones usually don’t match.
So I wonder, how do you compensate the drift?

I tried to compensate by setting the mp4box-fps just by 1 higher than fps, e.g. fps=23, mp4box=24, what tuned out as video is ahead of the mic-audio.
I tried comma-values in the config-file directly, like fps=23.4 (period) or 23,3 (comma), and mp4box=23,8 or 23.9, and many other pairs of values, but as long as only integer values are processed, I can not find a matching value to compensate the drift.

Or, is there any parameter to adjust for the process that converts h246 and audio to mp4 to minimize the drift?

I´m using Raspberry Pi3+ with BusterLite as basis for PiKrellCam.
May be I should downgrade?

Right now I´m out of ideas.
Any hint appreciated.

Greetings, Gunneg

billw2, 2019-08-01:
On Wed, 31 Jul 2019 14:17:54 -0700
AntonJouri <notifications@github.com> wrote:

> When recording video with audio, the mic is ahead of the video, and it is getting worse every minute.
> It is known, that the fps of Raspi-Camera and USB-Microphones usually don’t match.
> So I wonder, [...]

The config takes only integer values so there is no way to do any compensation now.  I could
change the type to float so factional values could be entered, but your asking this has made me think
that a better way would be for pikrellcam to calculate a fps needed to make video time match
audio time and use that in the mp4box.  I'm wondering about the magnitude of the drift you get, so
could you look in the pikrellcam log and tell me some rate results you get for 2 or 3 typical
video records?  Sound like you have long videos, but for a short video I get this for a "Motion record
stop" line in the log:

Motion record stop (size:5.0MB < /tmp:257.5MB)  vid_time:10.04  vid_fps:24.00  audio_frames:481965  audio_rate:
48000

If I could see some log numbers like that for some of your videos it would help and in the meantime I'll
think about making the changes to pikrellcam of fractional fps config and automatic calculating a mp4box fps.

Thanks,
Bill

-- 
Bill Wilson <billw@gkrellm.net>

AntonJouri, 2019-08-02:
Thank you very much for your quick response, Bill.

I made a clean install and 4 different recordings of 2 minutes each.

1) 24 fps  and 0 mp4box_fps
From the beginning, audio is a little faster than video, after 2 minutes audio is 2 seconds faster than the video.

2) 24 fps and 25 mp4box_fps
For the first 30 seconds it seems to be in sync, but then audio is slower than video, after 2 minutes 2 seconds behind.

3) 23 fps and 24 mp4box_fps (almost the same result)
For the first 30 seconds it seems to be in sync, but then audio is slower than video, after 2 minutes 2 seconds behind.

4) 23 fps and 22 mp4box_fps
After 2 minutes video is about 4 seconds faster than audio.

========================================================
======= PiKrellCam 4.3.0 started at 2019-08-02 17:18:47
========================================================
17:18:47 : Failed to open motion regions file: /home/pi/.pikrellcam/motion-regions.conf
17:18:47 :   execl[wait:1]: sudo chown pi.www-data /home/pi/pikrellcam/www
command FIFO: /home/pi/pikrellcam/www/FIFO
motion_detects FIFO  : /home/pi/pikrellcam/www/motion_detects_FIFO
audio FIFO  : /home/pi/pikrellcam/www/audio_FIFO
mjpeg stream: /run/pikrellcam/mjpeg.jpg
17:18:47 :   execl[wait:1]: sudo mkdir -p /home/pi/pikrellcam/media
17:18:47 :   execl[wait:1]: sudo chown pi.www-data /home/pi/pikrellcam/media
17:18:47 :   execl[wait:1]: sudo chmod 775 /home/pi/pikrellcam/media
17:18:47 :   execl[wait:1]: sudo mkdir -p /home/pi/pikrellcam/media/archive
17:18:47 :   execl[wait:1]: sudo chown pi.www-data /home/pi/pikrellcam/media/archive
17:18:47 :   execl[wait:1]: sudo chmod 775 /home/pi/pikrellcam/media/archive
17:18:47 :   execl[wait:1]: sudo mkdir -p /home/pi/pikrellcam/media/loop
17:18:47 :   execl[wait:1]: sudo chown pi.www-data /home/pi/pikrellcam/media/loop
17:18:47 :   execl[wait:1]: sudo chmod 775 /home/pi/pikrellcam/media/loop
17:18:47 :   execl[wait:1]: /home/pi/pikrellcam/scripts-dist/_init /home/pi/pikrellcam /home/pi/pikrellcam/media/archive /home/pi/pikrellcam/media /run/pikrellcam/mjpeg.jpg /home/pi/pikrellcam/www/FIFO /tmp/pikrellcam.log servos_off /home/pi/pikrellcam/media/loop
  making www/state link to /run/pikrellcam/state
  making www/mjpeg.jpg link to /run/pikrellcam/mjpeg.jpg
  www/media link is already set to /home/pi/pikrellcam/media
  www/archive link is already set to /home/pi/pikrellcam/media/archive
  making www/loop link to /home/pi/pikrellcam/media/loop
  /home/pi/pikrellcam/www/config.php: LOG_FILE not changed.
  /home/pi/pikrellcam/www/config.php: MJPEG_FILE not changed.
  /home/pi/pikrellcam/www/config.php: FIFO_FILE not changed.
  /home/pi/pikrellcam/www/config.php: PIKRELLCAM not changed.
  /home/pi/pikrellcam/www/config.php: SERVOS_ENABLE not changed.
  /home/pi/pikrellcam/www/config.php: VERSION updated to: 4.3.0
17:18:47 :   execl[wait:1]: /home/pi/pikrellcam/scripts/startup /home/pi/pikrellcam /home/pi/pikrellcam/media /home/pi/pikrellcam/media/archive /tmp/pikrellcam.log
17:18:47 :   execl[wait:1]: sudo chown pi.www-data /tmp/pikrellcam.log
17:18:47 :   execl[wait:1]: sudo chmod 664 /tmp/pikrellcam.log
17:18:47 :   execl[wait:1]: sudo mkdir -p /run/pikrellcam
17:18:47 :   execl[wait:1]: sudo chown pi.www-data /run/pikrellcam
17:18:47 :   execl[wait:1]: sudo chmod 775 /run/pikrellcam
17:18:47 :   execl[wait:1]: sudo mkdir -p /home/pi/pikrellcam/media/videos
17:18:47 :   execl[wait:1]: sudo chown pi.www-data /home/pi/pikrellcam/media/videos
17:18:47 :   execl[wait:1]: sudo chmod 775 /home/pi/pikrellcam/media/videos
17:18:47 :   execl[wait:1]: sudo mkdir -p /home/pi/pikrellcam/media/thumbs
17:18:47 :   execl[wait:1]: sudo chown pi.www-data /home/pi/pikrellcam/media/thumbs
17:18:47 :   execl[wait:1]: sudo chmod 775 /home/pi/pikrellcam/media/thumbs
17:18:47 :   execl[wait:1]: sudo mkdir -p /home/pi/pikrellcam/media/stills
17:18:47 :   execl[wait:1]: sudo chown pi.www-data /home/pi/pikrellcam/media/stills
17:18:47 :   execl[wait:1]: sudo chmod 775 /home/pi/pikrellcam/media/stills
17:18:47 :   execl[wait:1]: sudo mkdir -p /home/pi/pikrellcam/media/timelapse
17:18:47 :   execl[wait:1]: sudo chown pi.www-data /home/pi/pikrellcam/media/timelapse
17:18:47 :   execl[wait:1]: sudo chmod 775 /home/pi/pikrellcam/media/timelapse
17:18:47 :   execl[wait:1]: sudo mkdir -p /home/pi/pikrellcam/media/stills/.thumbs
17:18:47 :   execl[wait:1]: sudo chown pi.www-data /home/pi/pikrellcam/media/stills/.thumbs
17:18:47 :   execl[wait:1]: sudo chmod 775 /home/pi/pikrellcam/media/stills/.thumbs
17:18:47 :   execl[wait:1]: sudo mkdir -p /home/pi/pikrellcam/media/loop/videos
17:18:47 :   execl[wait:1]: sudo chown pi.www-data /home/pi/pikrellcam/media/loop/videos
17:18:47 :   execl[wait:1]: sudo chmod 775 /home/pi/pikrellcam/media/loop/videos
[... 59 lines cut ...]
Views: 301058
Re: PiKrellCam: motion vector detect + OSD web interface <https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=115583&p=968635&hilit=drift#p968635>
Do we know if NFS on the Pi has the performance required to handle live video recording? Has anyone shown that to work?

Regarding accurate timestamps, I have noticed that the "burned-in" date/time on the video occasionally has a hiccup when it doesn't update for a second or more. Also, on my cameras the nominally 24 fps is more accurately 23.04 fps. I measured this using an external strobe light and adjusting the rate for minimum drift.
[... cut ...]
USB Mikrofon, Gyvazla Omnidirektionaler Kondensator Lavalier Clip On Mic für Computer, Laptop, Podcast, Interviews, Netzwerksingen, Skype, MSN, Audio Video Recording [Plug & Play]

If you need as examples some videos with different settings that are out of sync, I can upload them on a server for you.

If it is a problem with the PI hardware and its internal clock going a little to slow or to fast, then may be your idea of a floating point ratio between fps and mp4box_fps might do the trick ;-)

Greetings, Gunneg

billw2, 2019-08-04:
Thanks for the data.  I need to spend some time working out if I can get pikrellcam to auto detect
and compensate for the audio drift.  But in the meantime, I just added changes for setting
fractional mp4box fps values and if you upgrade you can experiment with that.

Bill

On Fri, 02 Aug 2019 09:58:55 -0700
AntonJouri <notifications@github.com> wrote:

> Thank you very much for your quick response, Bill.
>
> I made a clean install and 4 different recordings of 2 minutes each.
>
> 1) 24 fps  and 0 mp4box_fps
> From the beginning, audio is a little faster [...]

AntonJouri, 2019-08-04:
Thanks, Bill.
I will do, a.s.a.p. on monday morning :-)
And I´m looking forward for the auto-detection-function.
Thanks again for your help an your great PiKrellCam.

Guntram

> Am 04.08.2019 um 17:42 schrieb billw2 <notifications@github.com>:
> 
> Thanks for the data. I need to spend some time working out if I can get pikrellcam to auto detect
> and compensate for the audio drift. But in the meantime, I just added changes for setting
> fractional mp4box fps values and if you upgrade you can experiment with that.
> 
> Bill

n1c085, 2020-02-24:
Hi,
I'm experiencing the same problem with the audio. Is the auto-detection already implemented? Can you give me some advice on how to set the fps and mp4box values to compensate the audio shift?

Here's the log output for a 42s video file: