billw2/pikrellcam #48: GUI responsiveness on Pi4b when video is set to 1080p

Sp4rkR4t, 2019-08-08:
I have several camera traps all using different Pi's and I've noticed on all that when video is set to 1080p the gui can be a bit slower but on the Pi4b it is much slower, for example to scroll through the menus I have to press a nav button wait a couple of seconds for the menu to move, then press another button, but if I stick it down to 720p it's smooth.

ristomatti, 2019-08-13:
I'm experincing this also. Although in my case the user interface is always slow but maybe the preview updates a bit faster on 720p.

Details:

Board: Raspberry Pi Model 4B 2GB
OS: Raspbian Buster Lite
Camera: Noir V2 camera / Chinese clone camera
Power: PoE HAT / USB-C with a 2.4A power supply
CPU temp: 55-65C

I tried the same two cameras on a Rasberry Pi Model 3B+ running an equivalent installation (installed both at the same time). It was as fast as previously. I guess this shouldn't have anything to do with `avconv` replaced with `ffmpeg` (949225d27557941177cb4734e873a424754f6759) as the problem was only with Model 4B?

I noticed that the OSD timestamp looks kind of blurry. Oh and both Pi 3B+ and 4B had a peculiar problem with the Noir V2 camera. The automatin image calibration did not seem to work at all - it was overexposed to almost white and had pink sections here and there. This might be an issue with the camera module though. Didn't yet try with another one. No such issues with the Chinese camera.

ristomatti, 2019-08-13:
I found an explanation regarding the pink NoIR camera picture. It's related to some new AWB algorithm:
https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=245994

I wonder if PiKrellCam supports toggling it on? Anyway this seems to be unrelated to this issue, I did not mean to sidetrack from the original Issue.

ristomatti, 2019-08-20:
To answer my own question, they "greyworld" WB mode is not listed as an option.