billw2/pikrellcam #51: upgrade to Bruster Raspbian

jefwu72, 2019-11-10:
Pikrellcam becomes slow in Raspberry Pi4 with Buster.   Is there a plan to make Pikrellcam compatible with ruster

dand11, 2019-12-03:
I am also experiencing very slow and laggy performance via the web GUI with Buster.  Was previously fine.  Is it a known issue?

billw2, 2019-12-04:
I don't see anything slow with Buster so I don't know.  But what is slow?  A better
description might help.

dand11, 2019-12-04:
Apologies for the rushed message previously.

I have done a completely fresh Buster image on a Pi 4.  When using resolution 1640x1242 the web GUI is very laggy - I have a timestamp on the image and it'll only update every 10 seconds or so.  Reducing the resolution to 1024x768 fixes the issue, but it was never an issue for me before.

I can iperf to the Pi at >30Mbit/s so it's not a connection thing.

I can also use raspivid to record a video at this (or higher) resolutions and the video records and plays back fine.  It seems to just be the web interface.

I used to use a Pi 3 with this same camera and it was fine, I then took the SD card out of my Pi 3 and put it into a Pi 4 and it was also fine.  I then did a brand new build onto the Pi 4, installed pikrellcam afresh and the web interface preview has been laggy ever since.

billw2, 2019-12-04:
I have a Pi 1 and Pi 2 with cameras running Buster and Pikrellcam and I have browsers
Firefox on a x86 desktop and Chromium on a Pi 4 desktop viewing the cameras.
With mjpeg_width set to 1640 (and mjpeg_quality 5) I don't see that lag.

On the Pi 1, I ssh'ed in and checked the update rate on the mjpeg image pikrellcam creates
and the web interface looks at:

pi@rpi5 ~$ cd /run/pikrellcam/

pi@rpi5: /run/pikrellcam$ for i in 0 1 2 3 4 5;do  ls --time-style='+%d-%m-%Y %H:%M:%S' -l | grep -w
"mjpeg.jpg$"; sleep 0.2; done -rw-r--r-- 1 pi pi 60718 04-12-2019 13:15:11 mjpeg.jpg
-rw-r--r-- 1 pi pi 60681 04-12-2019 13:15:13 mjpeg.jpg
-rw-r--r-- 1 pi pi 60681 04-12-2019 13:15:13 mjpeg.jpg
-rw-r--r-- 1 pi pi 59966 04-12-2019 13:15:13 mjpeg.jpg
-rw-r--r-- 1 pi pi 60745 04-12-2019 13:15:14 mjpeg.jpg
-rw-r--r-- 1 pi pi 60745 04-12-2019 13:15:14 mjpeg.jpg

I'm currently out of  town and the Pi 1 is at home connecting over wifi to a router and the Pi 4 with
Chromium I'm viewing on remotely is updating the image with about a 250KB/sec data stream.

So at least for the Buster setups I now have I'm not seeing a problem.  The case I haven't checked
is a Pi 4 with a camera running Buster and to do that test I will have to wait until I get back home.

Maybe you can check the mjpeg.jpg image update rate on your camera Pi as I did above.  If
that is updating as it should then for some reason the nginx web server is slow serving the
image.

dand11, 2019-12-04:
Thanks for your efforts bill.

I performed a similar test to you, with the following result:

`pi@pi:/run/pikrellcam $ while [ 1 ]; do  ls --time-style='+%d-%m-%Y %H:%M:%S' -l | grep -w "mjpeg.jpg$"; sleep 0.2; done`
`-rw-r--r-- 1 pi pi 13848 04-12-2019 20:31:21 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 13848 04-12-2019 20:31:21 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 13848 04-12-2019 20:31:21 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 13848 04-12-2019 20:31:21 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 13848 04-12-2019 20:31:21 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 13848 04-12-2019 20:31:21 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 13848 04-12-2019 20:31:21 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14317 04-12-2019 20:31:24 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14328 04-12-2019 20:31:33 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14328 04-12-2019 20:31:33 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14328 04-12-2019 20:31:33 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14328 04-12-2019 20:31:33 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14328 04-12-2019 20:31:33 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14328 04-12-2019 20:31:33 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14328 04-12-2019 20:31:33 mjpeg.jpg`
`-rw-r--r-- 1 pi pi 14328 04-12-2019 20:31:33 mjpeg.jpg`

As you can see, there are gaps between new images.

I'm using the following settings:
mjpeg_quality 8
mjpeg_divider 4

billw2, 2019-12-04:
So pikrellcam goes several seconds with no updates to that file.  Offhand I have no
explanation so I'll have to setup a Pi 4 with camera and see if I can reproduce the
problem.

Thanks for running the test.

dand11, 2019-12-18:
Any luck @billw2 ?

After a few days the video freezes permanently.  I've ruled out the camera now and done a complete reinstall, to no avail sadly.

mattjb1222, 2019-12-28:
@dand11 Have you ruled out the microsd as being the culprit? If you haven't tried it yet already, maybe set up /run/pikrellcam (and other frequently written filesystems) as a ramdisk like so (also increases lifetime of the microsd):

pkc@pi:~$ cat /etc/fstab
/dev/mmcblk0p1 /boot vfat defaults 0 2
/dev/mmcblk0p2 / f2fs noatime 0 0
tmpfs /tmp tmpfs defaults,nodev,nosuid 0 0
tmpfs	/var/log	tmpfs	defaults,noatime,nosuid,mode=0777,size=10m    0 0
tmpfs	/var/log/nginx	tmpfs	defaults,noatime,nosuid,mode=0777,size=5m    0 0
tmpfs	/run/pikrellcam	tmpfs	defaults,noatime,nosuid,mode=0777,size=50m    0 0

I use a minimal Buster image from https://github.com/FooDeas/raspberrypi-ua-netinst on a Raspberry Pi 3 model A without issue, but I don't have a RPi 4 to test with.

dand11, 2020-01-07:
No luck unfortunately.

I've got a brand new (decent) SD card, installed a fresh buster-light image onto my 2GB Pi 4, and only installed pikrellcam.  I see the same lag.  Who knows!

I'll just use it at a low resolution for now.

RawLiquid, 2020-02-25:
If it will help, I can provide you with remote access(shell/vnc) as well as some of my time if you need any assistance with testing(FaceTime or other video chat)

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

RawLiquid, 2020-02-25:
<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Biggregw, 2020-05-31:
Did you ever resolve this? I have been wondering about performance with my Pi4 as I'm sure my Pi3 never missed a beat.
After reading this I'm thinking of swapping the Pi4 with my Pi3.

dand11, 2020-05-31:
I didn’t I’m afraid. There’s certainly an issue with the pi 4. I’ve just resorted to using it in a low res mode

billw2, 2020-05-31:
On Sun, 31 May 2020 15:03:45 -0700
dand11 <notifications@github.com> wrote:

> I didn’t I’m afraid. There’s certainly an issue with the pi 4. I’ve just resorted to using it in a low res
> mode

The pi4 with HQ camera shows a web jpeg lags with even lower resolution.  Some limits in the GPU
pipeline are getting hit with converting jpegs along with recording continuous h264 video into the circular
buffer.  The jpeg callback just does not clear and so frames get dropped.

For the pi4 + HQ camera case I resorted to lowering video_fps to 20 and then no more web preview lagging.
It's not ideal but for now I don't know if there can be a better solution.

Biggregw, 2020-06-01:
Thanks for the tip re framerate which definitely improved things. 
Would there be any further improvement if I overclock the GPU?

billw2, 2020-06-02:
On Mon, 01 Jun 2020 14:23:36 -0700
Biggregw <notifications@github.com> wrote:

> Thanks for the tip re framerate which definitely improved things. 
> Would there be any further improvement if I overclock the GPU? 
> 

I have not tried overclocking, but it would be an interesting test.  I do find it strange
that there is this issue on a Pi 4 since the GPU is already clocked higher than previous Pi
versions.

So I also still need to try running an HQ camera on a Pi3 and Pi2.

Biggregw, 2020-06-02:
I had a go with one of my Pi2's and didn't really notice any difference.
Setting the fps to 20 helped. 
Overclocking the Pi4 didn't seem to make any difference nor did changing the gpu memory allocation to 512.
When I monitor the cpu usage it does look very low and the ram usage is also nowhere near max. 
Perhaps the GPU on the Pi4 isn't upto the higher resolutions. 
Shame as this is by far the best cctv software for the pi.

Blunderfoot, 2022-02-17:
Was this ever resolved. I have this same problem on a pi 3 on buster