billw2/pikrellcam #56: Good news a new camera module, will it work with our favorite pikrellcam software?

Yves911, 2020-04-30:
Hey guys,

Have you seen it already ? 
https://www.raspberrypi.org/products/raspberry-pi-high-quality-camera/

Looks interesting isn't it?
Cheers
Yves

Biggregw, 2020-05-31:
I'm using it with my Pi4 setup.
Works fine. Only issue is that Pikrellcam struggles with the higher resolution and drops fps on the stream so I can't really go full resolution.
Perhaps this is always going to be the case with any Pi software. 
Doesn't look to be using much of the cpu so not sure where the bottleneck is.

billw2, 2020-05-31:
On Sun, 31 May 2020 13:25:17 -0700
Biggregw <notifications@github.com> wrote:

> I'm using it with my Pi4 setup.
> Works fine. Only issue is that Pikrellcam struggles with the higher resolution and drops fps on the stream so
> I can't really go full resolution. Perhaps this is [...]

The GPU seems to have trouble keeping up with the two pipelines: the web page jpeg stream and
the video h264 recording.  What I have resorted to is reducing the video_fps to 20 and the jpeg stream
then keeps up.

The jpeg callback for the stream is stalling and I will try some things there, but at the least I'll put in
some diagnostics to make it clear that is what is happening.