billw2/pikrellcam #58: Timelapse configuration

bristolavon, 2020-06-24:
I've been using pikrellcam for a couple of years for wildlife photography and found it excellent for shooting videos.

I would  like to be able to use it to also take timelapse stills, specifically from when powered up to when powered down and at 30 second intervals.

I can set these paramaeters by connecting to the web page, but I can't find anyway of saving the settings so that 'it just works' like the video capture does. Is it possible, and if so, how, please?

Thank you.

Blimpyway, 2020-06-24:
for such a simple thing you can simply use a short script invoking raspistill at timed intervals. It can be launched repeatedly via crontab or as a init daemon

bristolavon, 2021-03-05:
Thanks for the help

closed 2021-03-05 by bristolavon