billw2/pikrellcam #59: Latest version breaks on jessie

jesscold, 2020-06-26:
Upgraded for 4.1.6 to latest and got this error when starting:
```bash
./pikrellcam: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.27' not found (required by ./pikrellcam)
./pikrellcam: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by ./pikrellcam)
```

The internets indicate that you cannot upgrade ldd using apt-get, so i got the thing working by building a new executable from source by running `make` in the `/src` dir.

This fixed it. Leaving this "issue" here for the next dweeb.

Egregius, 2020-07-04:
Thanks for this fix!
Got the same issue and this fixed it.

LucidEye, 2021-06-21:
Thank you.  This fixed the same issue for me as well.