billw2/pikrellcam #62: I've installed everything and cant click on anything

rcpilotp51, 2020-07-22:
I just got finished installing this on a raspberry pi 0 w and when going to the web interface nothing is responsive, meaning i click on something and nothing happens - especially with the image preview. is there anything further i need to do, thanks

~keith

rcpilotp51, 2020-07-22:
The image preview isnt available im using a pi0w and a picam v1.3 i can use raspistill to take an image, yet there us no preview in the interface. I cant click on any of the buttons, or they dont seem to do anything

billw2, 2020-07-22:
On Wed, 22 Jul 2020 05:31:04 -0700
Keith Brown <notifications@github.com> wrote:

> The image preview isnt available im using a pi0w and a picam v1.3 i can use raspistill to take an image, yet
> there us no preview in the interface. I cant click on any of the buttons, or they dont seem to do anything
> 

Is pikrellcam running?  At first install you need to open System and click "Start".  If you enabled auto start
in the install then pikrellcam will auto start on subsequent boots.

If there's still a problem, open a terminal and run "pikrellcam" and then "pikrellcam -v" to look at
possible error messages.  Also in the web interface do you see log lines if you click System->Log?

rcpilotp51, 2020-07-23:
Thanks i reinstalled using the user pi...my created user didnt work...thanks

rcpilotp51, 2020-07-23:
going to close this: others should do the following 

- install raspian (not lite)

- secure user "pi"
- update 
-configure camera and resolution
- install pikrellcam
- enjoy

closed 2020-07-23 by rcpilotp51