billw2/pikrellcam #65: rtsp instructions for gstreamer are outdated and not working

fotisK69, 2020-08-27:
Hi,

I tried to follow the instructions that are linked in the help page of pikrellcam at chapter Architecture, see "[rtsp live video setup instructions](https://www.raspberrypi.org/forums/viewtopic.php?p=862399#p862399)" but first they are outdated and second not at all working.

Setup:
- Raspberry pi Zero w + raspberry camera
- pikrellcam installed and up & running
- HTTP web page and stream working fine

```
root@raspberrypi:~# /root/gst-gateworks-apps/bin/gst-variable-rtsp-server -d 99 -p 8555 -m /stream.high -u "( tcpclientsrc port=3000 do-timestamp=true blocksize=262144 ! video/x-h264,stream-format=byte-stream,profile=high ! h264parse ! rtph264pay name=pay0 pt=96 )"
[1]:main:475 - set debug level to: 99
[1]:main:483 - set port to: 8555
[1]:main:479 - set mount point to: /stream.high
Usage: gst-variable-rtsp-server [OPTIONS]

Options:
 --help,            -? - This usage
 --version,         -v - Program Version: 1.4
 --debug,           -d - Debug Level (default: 0)
 --mount-point,     -m - What URI to mount (default: /stream)
 --port,            -p - Port to sink on (default: 9099)
 --steps,           -s - Steps to get to 'worst' quality (default: 5)
 --max-bitrate,     -b - Max bitrate cap, 0 == VBR (default: 10000)
 --min-bitrate,        - Min bitrate cap (default: 1)
 --msg-rate,        -r - Rate of messages displayed (default: 5s)

Examples:
 - Create RTSP server out of user created pipeline:
        gst-variable-rtsp-server "videotestsrc ! v4l2h264enc ! rtph264pay name=pay0 pt=96"
```

Please help with new updated instructions for using gstreamer with pikrellcam.

Thanks

bintravel, 2020-09-26:
I had the same problem, was caused by newer version of gst-gateworks in my case. You could (temporarily) solve this i think by using 'git checkout 96970c9b2596258863b057002154a80651540c2a ' after cloning into gst-gateworks-apps

troffasky, 2021-03-04:
If you omit the -u, then gst-variable-rtsp-server will run and accept connections. It doesn't actually work, however.
Doing a packet capture, I see:

GET /stream.high HTTP/1.1
Host: 192.168.1.25:8555
Accept: */*
Accept-Language: en_GB
User-Agent: VLC/3.0.11.1 LibVLC/3.0.11.1
Range: bytes=0-

HTTP/1.1 503 Service Unavailable
Server: GStreamer RTSP Server
Connection: close
Cache-Control: no-store
Pragma: no-cache
Date: Thu, 04 Mar 2021 22:34:18 GMT

troffasky, 2021-03-07:
Some slight progress here. Adding host=127.0.0.1 to the pipeline actually gets some video flowing! I think this might have been alluded to in a forum post about disabling IPv6. gst-variable-rtsp-server presumably tries localhost by default and localhost here is ::1. Unlike every other service, pikrellcam is only listening on IPv4, so it doesn't work, hence the 503.
That's the end of the good news, however. I get about 3-4s of video and then gst-variable-rtsp-server segfaults.