billw2/pikrellcam #72: Extreme red glow on my pi noir images/videos

hvdwolf, 2021-04-19:
Hi, I (finally) upgraded my pi zero W from Jessie to Buster (via Stretch), and also upgraded pikrellcam.
Everything works fine again, but I now have a extreme red/orange glow in my videos/stills using a V1.3 lens. As if I am on Mars (at least that's how they convince it to be on Mars).
I tried every image option in the Settings, but nothing changes.

I also tried this  lens on a RPi3 (fresh install with buster with latest version of pikrellcam) but it has the same red glow.
I tried another IR V1.3 lens from which "I remember" that it just had the "normal" IR colors, both on the zero W and on the RPi3. It has the same red/orange glow, instead of the "normal" IR colors.

Anyone an idea?

pageauc, 2021-04-19:
I normally test with raspistill to verify camera hardware.  Quite often it
is a ribbon cable connection issue and sometimes the camera LED is on and
reflects off something.  I also had a problem not realizing a camera module
was a noir version.

hvdwolf, 2021-04-19:
Thanks for your reply, but it really is an IR lens, and so is the other one I tried. I also have a normal V1.3 lens and a normal wide-angle lens. Both work fine on both my pi zero W and RPi3.
I connected the lens to my zero W, using that specific zero camera cable, and to my RPI3 using a normal camera cable. I have multiple normal camera cables and also tried another one on the RPi3  (sorry for not mentioning that). It doesn't make any difference. To me that rules out any cable issue.
I also have "pre-upgrade" and "post-upgrade" movies. The old ones are the normal IR washed out, strangely weird colors. The new ones are simply orange/red.

hvdwolf, 2021-04-20:
I found it.
At first I thought it had to do with swapping the red and blue channel, which is oftenly done in infrared photography. So I already tried that with ffmpeg on my videos, but that was not the reason and it didn't solve anything.
I found the following post: [New AWB algorithm - NoIR camera support](https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=245994). 
In short: Due to changes in the code no longer using the quad processors in the GPU, you need to compensate for this red/purple tinge. For that a new AWB mode is introduced: greyworld.
Choose Setup -> Camera Params -> "White Bal" and set it to greyworld for an InfraRed lens.
(It is one of 2 options I did not try as I thought it was to change to black&white)

closed 2021-04-24 by hvdwolf