billw2/pikrellcam #78: buster and pikrellcam

Tim-Barker, 2021-11-16:
I've been using pikrellcam for a while now and once it was going have had few problems with it, an excellent though oft complicated piece of software which worked well for what I wanted at the time (probably 5 years ago).

It was installed initially to monitor activity in our garage which was broken into from a 'common area' of a set of flats soon after we arrived and more recently for extended periods when we weren't present.   It served it's purpose well, at least until a couple of years ago.  As you might imagine we haven't been away much over the past 2 years partly as a result of the closure on International borders so I haven't been monitoring it or the camera for a while.  Now that things appear to moving again it was time to check everything and make sure it was working the way I wanted and at the same time bring the software up-to-date across the common Raspberry Pis I have been using.

So I tried using the system as it was and had no luck then I upgraded to the latest version of buster which worked ok for a couple of days then things started to go awry.  As far as I am aware I didn't do anything strange just prior to it not working but...

An explanation of the system:-
2 x Raspberry Pi running Raspbian buster `Linux xxxxxx 5.10.63+ #1459 Wed Oct 6 16:40:27 BST 2021 armv6l GNU/Linux`.  Image used being `Raspberry Pi OS Lite` and with a number of other pieces of software such as perl added.  pikrellcam installed as per instructions.  One of the Pi's is an original "B" model with no wireless and 2 USB sockets the other a more recent B+ both seem to work without obvious issues.

 ```
cd /home/pi
 git clone https://github.com/billw2/pikrellcam.git
 cd pikrellcam
 ./install-pikrellcam.sh
```

After running` install-pikrellcam.sh` and going to the page `http://168...115:8081` I now just get the error message "No preview...possible causes...".  When first tried the upgrade and the software was running properly I changed one or two file locations and permissions and it is these, I wonder, where the errors crept in or is it more than this?  From what I can see it all makes sense and 'should' work but then maybe there is something else going on.  I need to add that, given buster uses systemd, I did follow instructions elsewhere for adding a pikrellcam.service file into /etc/systemd/system as per the instructions elsewhere in the 'issues' section although slightly modified (VIZ Cyber1000's  "Error on autostart if /etc/rc.local is missing #68".  

My first question would be, what user:group permissions are required on all files/directories involved?  That may be all that is involved.

Some files from the system that might help pinpoint a problem are (all blank lines and comments extracted):-
~/.pikrellcam/pikrellcam.conf
```
install_dir /home/pi/pikrellcam
media_dir /var/www/camera_1
archive_dir archive
loop_dir loop
log_file /var/log/pikrellcam.log
log_lines 500
on_startup $C/startup $I $m $a $G
halt_enable on
multicast_enable on
loop_enable off
loop_record_time_limit 30
loop_diskusage_percent 30
motion_stills_enable off
motion_stills_per_minute 30
motion_enable on
motion_off_preset off
motion_record_time_limit 0
motion_vectors_dimming 45
motion_confirm_gap 4
motion_event_gap 20
motion_pre_capture 10
motion_post_capture 10
motion_preview_clean on
motion_show_counts off
preview_stall_warning on
motion_area_min_side 80
motion_stats off
motion_detects_fifo_enable off
audio_trigger_video off
audio_trigger_level 50
box_MP3_only off
video_motion_name_format motion_%F_%H.%M.%S_$N.mp4
video_manual_name_format manual_%F_%H.%M.%S_$N.mp4
diskfree_percent 20
check_media_diskfree off
check_archive_diskfree off
video_width 1
video_height 1080
video_fps 24
video_mp4box_fps 0.000
video_bitrate 6000000
mjpeg_width 1920
mjpeg_quality 80
mjpeg_divider 2
still_name_format image_%F_%H.%M.%S_$N.jpg
still_width 2592
still_height 1944
[... 248 lines cut ...]
...and for completion the /etc/systemd/system/pikrellcam.sh file consists
```
[Unit]
Description=Pikrellcam
After=sshd.service network-online.target

[Service]
User=video
group=video
Type=simple
ExecStart=/home/pi/pikrellcam/pikrellcam
ExecReload=/bin/kill -HUP $MAINPID
Restart=on-failure
RestartPreventExitStatus=255
RestartSec = 15
RuntimeDirectory=/home/pi/pikrellcam
RuntimeDirectoryMode=0755

[Install]
WantedBy=multi-user.target
Alias=pikrellcam.service
```
I think that is the correct setup for using pikrellcam under systemd, perhaps something is wrong here (as well).  Maybe there should be a user/group specified as well?

What am I missing?  Let me know if there are any other files of use and I will post them.

billw2, 2021-11-17:
Hi,

This is Jesse Wilson, Bill's brother.  I am monitoring his email accounts.

I'm sorry to inform you that Bill passed away on October 14th.

Unfortunately I don't know anything about his software so I can't help.

Assuming I can figure out where to post the information, I will eventually post something to his internet hang outs but that won't happen for a while.

Regards,

Jesse

Tim-Barker, 2021-11-17:
Dear Jesse

 

So sorry to hear about Bill and fully understand the comments.  I’m more than happy to see whether some comments come from others involved on Github so don’t worry about it.  I suspect it is a problem with the Operating system changing some underlying way the video drivers work and stopping this software from accessing the camera.  I’ll continue to look for solutions and post any I find.  May I put a comment on Github mentioning Bill’s passing and asking if there is anybody interested in taking over the development?

 

Regards

Tim Barker.

billw2, 2021-11-18:
Hi Tim,

Yes, please post to github if you wouldn't mind.  I've written an obituary which is below.  I don't know if it would be appropriate to post it to github or not.  If you think it is ok to post an obituary, then please feel free to use the obituary as is or modify it if you think it is too long.

Thank you for your help!

Regards,

Jesse
-------------------
William (Bill) James Wilson, Jr., passed away on October 14, 2021 in Kyle, Texas, after a short bout with cancer.

Bill was born on July 4th, 1953 in Baytown, Texas.  Although born in Baytown, Bill spent the first 13 years of his life living in Venezuela where his father worked for Creole Petroleum Corporation.  The family moved several times while in Venezuela, spending time in small, out of the way oil camps such as Caripito and Quiriquire.  In 1966, Venezuela nationalized the oil industry and booted all foreign nationals out of the country.  The family returned to Texas to live in Crosby for 2 years before permanently settling in Corpus Christi where Bill attended W.B. Ray High School and graduated as the class Salutatorian in 1971.  He then left for college at Rice University in Houston, graduating in 1975 with a BSEE and in 1976 with an MEE.  After graduation, Bill spent a few years working at Texas Instruments in Houston.  While at TI, he became interested in the stock market and soon realized investing was the path he wanted to follow as it would allow him to be independent and be his own boss.  After leaving TI, he never looked back and was his own boss as a successful investor for almost 40 years.  As an investor, Bill pioneered the work from home concept starting in the mid-80’s, long before it recently became popular.

Bill was an incredible engineer.  He put his Rice education to good use, first, at Texas Instruments and then afterward when he was always tinkering with electronics and writing software.  Bill had an insatiable curiosity which moved him to figure out how things worked.  He built from scratch stereos and speakers, radio receivers, telescopes (with home ground mirrors), and computers which used his homegrown UNIX based OS.  Bill also developed and supported a couple of software open source projects - GKrellM, a computer system monitor, and PiKrellCam, a video security system.

Bill very much enjoyed working with his hands.  After years of living in Houston and Friendswood, Texas, he bought land in Buda, Texas, where he eventually built a house into which he moved in 2014.  He did most of the interior work himself.  He laid all the floor and bathroom tile, finished the plumbing and interior trim, made all the kitchen and bathroom cabinets including the granite countertops, and installed all the interior lights.  He especially enjoyed woodworking.  He mastered techniques for putting beautiful finishes on his work and sometimes surprised family and friends with gifts of furniture.

Growing up, Bill always said all the fireworks on July 4th were a celebration of his birthday.  Bill was also a bit of a practical joker, especially around Christmas time.  He loved to make it a challenge for his gift recipients to figure out which gift was theirs.  He would often mislabel presents or even not label them at all.  He would sometimes also make it difficult to unwrap presents through copious use of duct tape and other difficult unwrapping methods.

Bill was fiercely loyal and dependable.  His family and friends could always count on him to help no matter the problem.  He would always show up with a trunk full of tools ready to solve problems or just help.

Bill was preceded in passing by his parents, William James Wilson, Sr. and Ora Faye Lostak Wilson.  He never married and is survived by his family of three siblings: Mary Anne Smith, Carolyn Wilson, and Jesse Wilson and wife Jackie and their kids Marshall, Madeleine, and Cornel; numerous cousins; and his very dear friends Betty Tillman, Josh and Traci Tillman, and Sara Jur.  Bill’s untimely passing has left a very large hole in all our hearts.  He will be dearly missed.

In remembrance of Bill, please make a contribution to the American Cancer Society in Bill’s name.
-------------------

billw2, 2021-11-18:
Woops, that last sentence wasn't supposed to be "bolded".  Please fix that before you post it.  Thanks!