billw2/pikrellcam #80: install script for OS Bullseye

rleyden559, 2022-02-16:
I assume noone is now updating the install script and I noticed that it does not work , as is, under Bullseye.
I found that if you edit the script to change all instances of php7.3 to php7.4 (three times getting apps and once setting up nginx), the script and programs seems to work (for now). Of course, activate legacy camera in raspi-config.

kajan001, 2022-05-30:
Thanks  a lot for the update Bro!

davidolmos, 2022-10-02:
I am running Bullseye (64 bit) in a Raspberry Pi 3 and I get the following error message when running the installation script: 
```
./install-pikrellcam.sh: line 194: ./pikrellcam: No such file or directory
Unexpected failure to create config file /home/david/.pikrellcam/pikrellcam.conf
```
 I tried both to run it as is in Github and as well following rleyden559 comment above to amend the script from "php7.3" to "php7.4". In both cases I get the same message.

Thanks for helping.

davidolmos, 2022-10-02:
Just fixed it by creating the directory .pikrellcam and an empty file pikrellcam.conf (with touch pikrellcam.conf command) in my home directory (david)

kajan001, 2023-03-21:
I am not sure bullseye will work but a legacy buster version still works. 
check out issues 80 & 82
https://github.com/billw2/pikrellcam/issues/82
https://github.com/billw2/pikrellcam/issues/80#issue-1139443042