billw2/pikrellcam #81: install success but no pikrellcam webpage

jhemberger, 2022-03-02:
I am installing pikrellcam onto a Raspberry Pi Zero W. Successful installation using default settings, however I am unable to successfully load the pikrellcam webpage either as: "http://your_pi" or replacing your_pi with my pi's hostname.

kajan001, 2022-03-04:
Pikrellcam is no longer updating because the designer died late 2021.

There are a possible solution in the last posts on this forum https://forums.raspberrypi.com/viewtopic.php?p=1965930#p1965930

kajan001, 2022-03-04:
also here https://github.com/billw2/pikrellcam/issues/80

jhemberger, 2022-03-04:
Sorry to hear that! He seems to have been a big figure in the pi world. I edited the install script per #80 and that seems to have worked! Thanks for the redirect on that.

closed 2022-03-04 by jhemberger