billw2/pikrellcam #82: Will someone carry on the legacy?

kajan001, 2022-03-30:
Is there anyone here that is prepared to carry on the legacy of Bill in Pikrellcam?
I would gladly do it myself but this is way over my paygrade.

Waltertscott10, 2022-04-03:
I do hope someone can step up, this is a great piece of software,
I certainly would if I had any idea what I was doing, it has taken me two days to figure out how to modify the install script for bullseye.
I'm no programmer just it just a hobby for me would love to mess around and learn the code but don't have time or knowledge for maintaining it.

kajan001, 2022-05-30:
Anyone? Please!!

lucaji, 2022-10-29:
I updated the install and python scripts, updated jQuery and Bootstrap to latest version thus modernizing parts of the web interface as well.

There is a brand new proto-dashboard for sensor logging (temp and humidity).

I believe to have made too many substantial changes, and being a rough (and unfinished yet) implementation from my side at the time of writing, I won't spawn a PR to sync with this original repo, but please accept this note as heads up for anyone interested.

My repo available [here](https://github.com/lucaji/pikrellcam).

Any thoughts, advice, comments are welcome.

Tristan79, 2022-10-31:
+1 upvote you take over

gbmaple, 2023-03-24:
I started to use this great piece of software, a couple of years ago and I am really impressed with it. There is so much built in functionality and it is very easy to install and maintain. 
Short term, the install script needs to be modified to support Bullseye which uses php 7.4, also since Bullseye has moved from picamera to libcamera, you will need to activate legacy camera support in Raspi-config.
Long term, pikrellcam will need to be switched from using picamera to libcamera and then Bullseye 64 bit support (perhaps?).

kajan001, 2023-03-24:
I have used the buster legacy (2022 something) and pikrellcam and that still works. I have been trying to update the installer so it works with bullseye and other debian kernels (tinker board) but I do not have the know-how.
Does anyone  know how to work the main pikrell cam program? I think it is in c or c++. I have not even been able to open it.

Can we make this a sticky?

rleyden559, 2023-04-04:
My hope would be that pikrellcam could be modified to use libcamera or libcamera2 to enable to use of newer cameras, including Arducam cameras.  I have purchased 4 different Arducam cameras, the most useful to me being the low light camera.  All of these require Bullseye and libcamera (plus perhaps  Arducam libraries).
My guess is that would require such a major re-write that even calling it a fork would be a stretch.
The fork  [lucaji](https://github.com/lucaji) is welcome.  I had been looking for a way to display temperature.
I've written simple python programs to use my Arducam's.  Capturing images on motion or pir is easy.  Streaming using libcamera seemed flaky the last time I looked into it (Summer 2022)

kajan001, 2023-04-05:
I have tried to install pikrellcam on debian 10 (tinkerboard with picam v2) but with no success. I am guessing the libraries attached to the camera are not the same and since I know nothing of how c programing works I dont know how and where the c-program gets the stream and vectors from the camera. Otherwise maybe these could be changed. 

A guess would be in pikrellcam.h though:

gbmaple, 2023-04-06:
I looked into libcamera and the interface examples (e.g. simple-cam) are
programmed in C++.
Bill wrote Pikrellcam in C, so to integrate with libcamera Bills code would
need to be reprogrammed using C++.
Thankfully, many C datastructures and functions can be used directly in C++
and C++ is object oriented.
I think we should maintain Bills great code, including the UI and just
change the interface to libcamera.
I know Python and a little C. So ideally we would need help from some C++
programers. Once built we could then test using are various hardware and
camera setups.

kajan001, 2023-04-09:
Is libcamera the equivalent of the mmal lib used in previous releases? 
A quick Google makes it seem possible to call c++ Library from a c program.

emjas92, 2024-09-10:
You may be interested to see that RPi_Web_Cam_Interface, which has the same legacy camera limitation, has a team making a Python 'backend' for the web interface that uses libcamera:
https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/688#issuecomment-2340955170

They explain the pros and cons in their presentation.
I'm not sure that it's the right approach for either camera project, but it's good to see some progress in this space.

_(I used RCWI for a while before switching to Pikrellcam and still keep up with the activity.)_