billw2/pikrellcam #84: Which OS version to choose for a fresh install?

CharlesButcher, 2022-07-09:
I am struggling to get pikrellcam working on a Pi Zero 2 W. It was previously installed on this same Pi and ran without problems, but now the web preview doesn't work properly and the system freezes. Can anyone recommend the best Raspberry OS (or any other OS) version to start from?

I've tried Bullseye Lite with [@rleyden559’s tweak](https://github.com/billw2/pikrellcam/issues/80), the current legacy version of Buster Lite, and an archived version of Buster Lite from 2020. In each case the pikrellcam install seemed to go OK and the web server started, but some of the buttons on the web page didn't work, the video froze, and the whole thing soon crashed.

This is a shame because the same hardware has been running pikrellcam very happily for a couple of years. I don't remember having any trouble with the original install. But last week I was messing around with a USB Wi-Fi adaptor, it didn't go well, and I decided to do a clean install – which I am now regretting.

I don't have the energy to do a lot of troubleshooting, but if anyone could confirm (or deny) that pikrellcam should work on a Pi Zero with the latest version of Buster, I'd be grateful.

Jimbolaya64, 2022-07-09:
I’m running on a zero 1 w with an up to date version of buster. No problems here. Maybe your SD card has failed?

CharlesButcher, 2022-07-09:
Thank you @Jimbolaya64. Just the kind of comment I was looking for. I've tried two SD cards, but both are old. I'll get a new one.

CharlesButcher, 2022-07-12:
Solved. It was [this old problem](https://github.com/billw2/pikrellcam/issues/45), fixed by adding `over_voltage=6` to `/boot/config.txt`. I'd had to do that to stop crashes on this same Pi Zero when setting up PiKrellCam two years ago. I made notes at the time, but forgot to read them. Oddly enough, my other Pi Zero didn't show that problem.

I stuck with the last release of Buster and all seems fine.