billw2/pikrellcam #87: Installation Fails

CycleMark, 2022-10-22:
Hi,

I've been using pikrelcam for years - but most recent update with latest download of Raspberry Pi OS is failing:

Set the port for the nginx web server.
If you already have a web server configuration using the default
port 80, you should enter an alternate port for PiKrellCam.
Otherwise you can use the default port 80 or an alternate as you wish.
The port number will be set in: /etc/nginx.sites-available/pikrellcam.
Enter web server port: 

For auto starting at boot, a PiKrellCam start command must be in rc.local.
If you don't start at boot, PiKrellCam can always be started and stopped
from the web page.
Do you want PiKrellCam to be auto started at boot? (yes/no): no

Enter a password for a web page login for user: pi
Enter a blank entry if you do not want the password login.
Enter password: 

Starting PiKrellCam install...
Installing packages:  php7.3 php7.3-common php7.3-fpm gpac nginx sshpass mpack imagemagick apache2-utils libasound2-dev libmp3lame-dev
Running: apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:3 https://packages.cloud.google.com/apt coral-edgetpu-stable InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'php7.3-common' for regex 'php7.3'
Package php7.3-common is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'php7.3-common' has no installation candidate
E: Unable to locate package php7.3-fpm
E: Couldn't find any package by glob 'php7.3-fpm'
E: Couldn't find any package by regex 'php7.3-fpm'
Removing pikrellcam autostart line from /etc/rc.local.
Adding to /etc/sudoers.d: www-data permission to run pikrellcam as user pi:
./install-pikrellcam.sh: line 285: htpasswd: command not found
chown: cannot access 'www/.htpasswd': No such file or directory
grep: /etc/nginx/nginx.conf: No such file or directory
Turning off nginx access_log.
sed: can't read /etc/nginx/nginx.conf: No such file or directory
Installing /etc/nginx/sites-available/pikrellcam
    nginx web server port: 1067
    nginx web server root: /home/pi/pikrellcam/www
cp: cannot create regular file '/etc/nginx/sites-available/pikrellcam': No such file or directory
sed: can't read /etc/nginx/sites-available/pikrellcam: No such file or directory
sed: can't read /etc/nginx/sites-available/pikrellcam: No such file or directory
Adding /etc/nginx/sites-enabled/pikrellcam link to sites-available/pikrellcam.
ln: failed to create symbolic link '/etc/nginx/sites-enabled/pikrellcam': No such file or directory
A password for the web page is not set.
sed: can't read /etc/nginx/sites-available/pikrellcam: No such file or directory
Failed to restart nginx.service: Unit nginx.service not found.

What's up?

Thanks

Mark

CycleMark, 2022-10-24:
Turns out it's the same php7.3 ->pho7.4 issue as mentioned here - https://github.com/billw2/pikrellcam/issues/80#issue-1139443042

kajan001, 2023-03-21:
I am not sure bullseye will work but a legacy buster version still works. 
check out issues 80 & 82
https://github.com/billw2/pikrellcam/issues/82
https://github.com/billw2/pikrellcam/issues/80#issue-1139443042