billw2/pikrellcam #88: New Install Won't Run

micksulley, 2023-01-19:
I've been using PiKrellCam for many years but my system has failed so I wanted to build a new one.
I found that when I ran the install script it failed, an except - 
```
Note, selecting 'php7.3-common' for regex 'php7.3'
Note, selecting 'php7.3-sass' for regex 'php7.3'
Note, selecting 'php-sass' instead of 'php7.3-sass'
Package php7.3-common is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'php7.3-common' has no installation candidate
E: Unable to locate package php7.3-fpm
E: Couldn't find any package by glob 'php7.3-fpm'
E: Couldn't find any package by regex 'php7.3-fpm'
```
I edited the install file and changed references to php7.3 to just php.  The install script now runs without errors, however I still get "Unable to connect" on the web page.
I am running Buster on a Pi3
Any idea how to fix?

Jimbolaya64, 2023-01-19:
Does this help? [https://github.com/billw2/pikrellcam/issues/87#issue-1419475359](url)

micksulley, 2023-01-19:
That doesn't help.  I edited the install file to remove the version references for PHP and the install runs fine, or seems to, but I still get "Unable to connect" on the web page.
This is my install -
```
pi@pi-porch:~/pikrellcam $ ./install-pikrellcam.sh

Set the port for the nginx web server.
If you already have a web server configuration using the default
port 80, you should enter an alternate port for PiKrellCam.
Otherwise you can use the default port 80 or an alternate as you wish.
The port number will be set in: /etc/nginx.sites-available/pikrellcam.
Enter web server port: 

For auto starting at boot, a PiKrellCam start command must be in rc.local.
If you don't start at boot, PiKrellCam can always be started and stopped
from the web page.
Do you want PiKrellCam to be auto started at boot? (yes/no): yes

Enter a password for a web page login for user: pi
Enter a blank entry if you do not want the password login.
Enter password: 

Starting PiKrellCam install...
No packages need to be installed.
./install-pikrellcam.sh: line 194: ./pikrellcam: No such file or directory
Setting install_dir config line in /home/pi/.pikrellcam/pikrellcam.conf:
install_dir /home/pi/pikrellcam
Adding a pikrellcam autostart command to /etc/rc.local:
su pi -c '(sleep 5; /home/pi/pikrellcam/pikrellcam) &'
Adding to /etc/sudoers.d: www-data permission to run pikrellcam as user pi:
Installing /etc/nginx/sites-available/pikrellcam
    nginx web server port: 80
    nginx web server root: /home/pi/pikrellcam/www
A password for the web page is not set.

Install finished.
This install script does not automatically start pikrellcam.
To start pikrellcam, open a browser page to:
    http://your_pi
and click on the "System" panel and then the "Start PiKrellCam" button.
PiKrellCam can also be run from a Pi terminal for testing purposes.
Automatic pikrellcam starting at boot is enabled.

```

micksulley, 2023-01-19:
Just realised I am running Bullseye, not Buster, don't know if that makes any difference

micksulley, 2023-01-24:
Having investigated and tested some more, I now have PiKrellCam running on a fresh install of Buster.  I cannot get it to run on Bullseye, my guess is that it is due to the changed to the camera software in the new release.  
Anyone know if PiKrellCam is likely to be upgraded to run with Bullseye?

AratoArpad, 2023-01-27:
> Having investigated and tested some more, I now have PiKrellCam running on a fresh install of Buster. I cannot get it to run on Bullseye, my guess is that it is due to the changed to the camera software in the new release. Anyone know if PiKrellCam is likely to be upgraded to run with Bullseye?

Hey,

I was on the same journey as you. I couldn't get it working, same with motioneye, the ribbon cable camera modules don't really cooperate well with the latest release. 

It's unlikely that PiKrellCam will get an official update for Bullseye. 
If I'm not mistaken, the creator of the project is not with us any more.

z-noob115, 2023-02-01:
micksulley, did you still change php. on the working fresh install? I'm striking out on every attempt

z-noob115, 2023-02-01:
thanks! I'll give it another go but my main problem is the home page is jumpy* and will not show any preview. Tried on Buster lite on RPI4 and Zero with same results

micksulley, 2023-02-01:
> micksulley, did you still change php. on the working fresh install? I'm striking out on every attempt

I changed php on the Bullseye install to get PiKrellCam to install at all, but then it didn't work.  On Buster it installed and worked without any changes.

micksulley, 2023-02-01:
I have seen that when the camera is not detected.  If you run 
raspistill -o testshot.jpg
can you open testshot.jpg?  If not it seems you have a camera problem.

z-noob115, 2023-02-03:
Thanks micksulley! Still have not figured out what i did wrong but have since gotten multiple instances working and doing some fine tuning. Still learning but the only thing i did different is turning off wifi on install image of OS. Dont see how that would make a difference, but I may be wrong.

fingerprince1, 2023-02-15:
> Having investigated and tested some more, I now have PiKrellCam running on a fresh install of Buster. I cannot get it to run on Bullseye, my guess is that it is due to the changed to the camera software in the new release. Anyone know if PiKrellCam is likely to be upgraded to run with Bullseye?

Check out the updates of user Lucaji, he seems to keep this software alive

jsn-0, 2023-02-21:
Have you tried enabling the legacy camera stack when running on Bullseye?

This thread has some details on libcamera replacing the old camera stack in Bullseye and how to re-enable the old legacy camera functionality:
https://forums.raspberrypi.com/viewtopic.php?t=323390

kajan001, 2023-03-21:
I am not sure bullseye will work but a legacy buster version still works. 
check out issues 80 & 82
https://github.com/billw2/pikrellcam/issues/82
https://github.com/billw2/pikrellcam/issues/80#issue-1139443042

Whyzanet, 2023-06-26:
I have pikrellcam running under bullseye and while it was some time ago I set it up, I have just updated it to the latest software without issues on the rpi3+ ( Kernel and OS 32 bit ).
IIRC I had to tweak it to look for php7.4 on the install, and enable the legacy camera stack as mentioned above.

Though it is a pity new camera hardware that will not work in legacy mode, will not work with pikrellcam as a result.

zweistein22, 2024-05-26:
Bullseye does not work any more.
Buster 32bit is the latest install where it works.
If interested, here a  [fork](https://github.com/zweistein22/pikrellcam) with correct install on buster (and -unneccessary php7.4 install and build from source integrated.
pikrellcam has a very nice feature which is the -coarse- video motion vectors are produced by the video hardware as a by product.  
The later libcamera and picamera2 (from bullseye onwards) do not offer that feature any more . This would require a full rewrite of bill's pikrellcam having to drop the nicest feature it has.  
Andreas

emjas92, 2024-09-10:
This Python 'backend' for RCWI may be of interest: https://github.com/billw2/pikrellcam/issues/82#issuecomment-2342034886

joedupa, 2025-07-27:
Bullseye works, you just have to make sure to download the 32 bit version of Raspberry Pi Lite OS and install it.
Then run this code to get rid of PHP version problems:

https://packages.sury.org/php/README.txt

and proceed with the installation.

After installation completes successfully,  make sure to enable legacy camera support with raspi-config -> interface options

Joe